SpinalHDL/VexRiscv #342: How to add a custom instruction that can read rd

LongStudy, 2023-05-10:
Hi, VexRiscv and SpinalHDL are really cool. I never thought adding SIMD instructions would be so easy.

As the example mentioned in the `CustomInstruction.scala`, the plugin can read `rs1` and `rs2` and write to `rd`. But I am looking for a way to read `rd` at the same time as shown below.

```
rd(7 downto 0) := rs1(7 downto 0) + rs2(7 downto 0) + rd(7 downto 0)
......
rd(31 downto 24) := rs1(31 downto 24) + rs2(31 downto 24) + rd(31 downto 24)
```

I learned that there is no part of RV32IMC instructions need to read `rd`, but I found some instructions in the p-spec that may require reading `rd`.

How can I add a custom instruction set that can read `rd`? (Of course, this instruction could be executed in two separate instructions, but perhaps one would be faster?) Thank you!

Dolu1990, 2023-05-10:
Hi,

That's currently not supported.
The best way would be to add support for a "RS3" in the RegFilePlugin and hazard plugin, i guess that should be enough ?

LongStudy, 2023-05-11:
Awesome, the efficiency of SpinalHDL is impressive. I modified `RegFilePlugin.scala` and `HazardSimplePlugin.scala` plugins and then modified the test cases in `crt.S` and run the regression test. The test results show that the rd register can be read efficiently.
Thanks a lot, VexRiscv is a great platform for me with so much to learn.

Dolu1990, 2023-05-11:
:D :D
Nice !

Let's me know if you push your code somewere, could be usefull ^^

LongStudy, 2023-05-11:
I am still learning and have no plan to push it to the Internet immediately. Thank you very much for taking the time to reply to the message. I think I have no problem for the time being, and maybe this issue can be closed.

LongStudy, 2023-10-31:
When I try to design custom instructions, I find that some complex instructions may consume more time after FPGA synthesis and cause timing violation. Can I imitate MulPlugin to put part of the logic into the memory stage?

Dolu1990, 2023-11-02:
Sure you can, no worries to have about it.

LongStudy, 2023-11-12:
Thanks a lot. I fixed the timing violation problem.

closed 2023-11-12 by LongStudy

ashuthosh-mr, 2024-03-01:
> Awesome, the efficiency of SpinalHDL is impressive. I modified `RegFilePlugin.scala` and `HazardSimplePlugin.scala` plugins and then modified the test cases in `crt.S` and run the regression test. [...]

Can you please share the changes that were made? This change might be the key to introduce MAC instruction. @LongStudy 

Thank you :)