SpinalHDL/VexRiscv #343: X over-propagation

flaviens, 2023-05-12:
Hi there!

## Problem description

Are you exclusively using Verilator to simulate VexRiscv?
I've been recently trying to simulate VexRiscv with Modelsim.

For some non-trivial cases such as the example below, the architectural registers become X. Is that a known problem? Is there any easy solution to that?

Other designs such as CVA6, Rocket and BOOM do not seem to have such problems.

## Example instance

Here is a waveform and the corresponding ELF file.
[example_instance.zip](https://github.com/SpinalHDL/VexRiscv/files/11464879/example_instance.zip)
The first X contamination to the integer register file happens at `PC = 0x800448e8` (the integer RF is architecturally never X before that, apparently).
You can then see the data being stored into memory as X close to the end of the simulation, which is the store instruction at `PC = 0x80026434`.

Thanks!
Flavien

Dolu1990, 2023-05-12:
Hi ^^

> Are you exclusively using Verilator to simulate VexRiscv

Yes right 

> For some non-trivial cases such as the example below, the architectural registers become X. Is that a known problem?

Not realy a known issue, but not explored case either XD

>  Is there any easy solution to that?

Hmm not that i know. What CPU configuration do you use  exactly ?

> Example instance

Ahhh in the VCD you sent me, there is only a few signals of the top ?
Would it be possible to have a full wave of the CPU signals ?

Thanks :D

flaviens, 2023-05-12:
Ah yes indeed, sorry. Was also noticing that it was quite lightweight :) . Now the VCD is huge! Let me get a shorter instance and come back to you shortly.

flaviens, 2023-05-12:
I updated the archive above with a much shorter execution (but the same problem probably), and with the full VCD this time :)
Thanks!

Dolu1990, 2023-05-12:
So, it seems like it is a xprop false positive issue.

I hate xprop very much <3, i love verilator very much too <3

So, the case we have here is that each time the CPU come to fetch a i$ cache line which isn't loaded, the i$ provide a 'X', as the i$ banks were never written before, that's to be expected. the issue is that those X messup with the main pipeline value bypass (RS1 / RS2) and make the things corrupt.

So, i would say, would need to force the initialisation of all the VexRiscv internal memories to be sure not hitting those cases.

There is a few things which can be enabled to force it (see https://github.com/SpinalHDL/VexRiscv/blob/c52433575dec04f10063b2fd7cebd0545c8b1be9/src/main/scala/vexriscv/ip/InstructionCache.scala#L318)

But that's not enough.
Will add it ^^

flaviens, 2023-05-12:
Hi @Dolu1990 thanks for looking into it! Extremely appreciated. I can help you validating your fix, just let me know.
Thanks!

Dolu1990, 2023-05-16:
Hi ^^

Here is a workaround : 

```` scala
  import spinal.core.internals._
  import scala.util.Random
  class PhaseRandomizedMem() extends PhaseNetlist {
    override def impl(pc: PhaseContext): Unit = {
      pc.walkDeclarations{
        case mem : Mem[_] if mem.initialContent == null => {
          mem.initBigInt(Array.fill(mem.wordCount)(BigInt.apply(mem.width, Random)))
        }
        case _ =>
      }
    }
  }

  val gen = SpinalConfig(inlineRom = true)
  gen.addTransformationPhase(new PhaseRandomizedMem)
  gen.generateVerilog(new VexRiscv(...))
```

this will ensure that all the memories in the design are initialized with a fixed random content

flaviens, 2023-05-16:
Thanks! Currently my generation flow looks very much like in [this demo](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/VexRiscvAhbLite3.scala), and then I just run `runMain vexriscv.demo.VexRiscvAhbLite3`. How would you integrate the workaround in this flow?
Thanks!
Flavien

flaviens, 2023-05-16:
Hi @Dolu1990,

Actually I closed the issue a bit fast. Actually the problem is still here :cry:  , even for the exact same ELF.  I copied exactly the workaround above and tried multiple times to be sure.

Dolu1990, 2023-05-16:
Hi ^^

Can you send the wave file with the updated design ?

Thanks :)

Dolu1990, 2023-05-16:
XD
No worries.

Where are the X ?  Can't find them ^^

flaviens, 2023-05-16:
I need some vacationmaybe... I used Verilator for the VCD :facepalm: You'll find the correct one replaced in place above (problem still here).
Thanks!

Dolu1990, 2023-05-16:
No worries ^^ 

In [sim_new.zip](https://github.com/SpinalHDL/VexRiscv/files/11488058/sim_new.zip) ? Still the same

flaviens, 2023-05-16:
Had not clicked "update comment". This issue is a festival :D

flaviens, 2023-05-16:
Looks good! Thanks!!

closed 2023-05-16 by flaviens