SpinalHDL/VexRiscv #347: Briey on AHBL

AdDraw, 2023-06-09:
Is it possible to change Briey's AXI crossbar to AHBL easily?

 I have tried a couple of times but all I get are errors.

Dolu1990, 2023-06-12:
Hi,

You can try to set both : 
- dBusCmdSlavePipe
- dBusRspSlavePipe
in the DBusCachedPlugin, this will isolate it.

https://github.com/SpinalHDL/VexRiscv/blob/c52433575dec04f10063b2fd7cebd0545c8b1be9/src/main/scala/vexriscv/plugin/DBusCachedPlugin.scala#L52

Should goes better then i think ^^

AdDraw, 2023-06-12:
As far as I can tell, `DBusCachedPlugin`'s dBus of type `DataCacheMemBus` does not support `toAxiLite3Master()` or any other `AxiLite3` type conversions.

Dolu1990, 2023-06-12:
Hooo ok ^^
Then you will need to add those pipelines in some other ways
Murax kind of do that : 
https://github.com/SpinalHDL/VexRiscv/blob/c52433575dec04f10063b2fd7cebd0545c8b1be9/doc/gcdPeripheral/src/main/scala/vexriscv/demo/Murax.scala#L266

AdDraw, 2023-06-12:
I have applied this to both IBus and DBus I can compile it now, thanks

What would be the best way to test that it still works? `src/test/cpp/briey`?

Dolu1990, 2023-06-12:
So, normaly, you can avoid having it on IBus, in order to preserve the performance.

> What would be the best way to test that it still works? src/test/cpp/briey?

Hmm there is nothing automated for the SoC level testing. So, i guess as long as it can run some software (ex dhrystone) and doesn't catch fire that's ok.

> I can close this now, @Dolu1990 would there be any merit in opening a PR for this?

I think yes, a function could be added to DBusSimpleBus, which would do your : 

```scala
          val dBustmp = new DBusSimpleBus
          dBustmp.cmd << plugin.dBus.cmd.halfPipe()
          dBustmp.rsp <> plugin.dBus.rsp
```
and return the dBustmp, that way we can just inline it.

Anyways, thanks for the feed back :)

closed 2023-06-12 by Dolu1990