SpinalHDL/VexRiscv #351: How can with Murax soc connect a memory on apb bus?

MartinaBarreiroGuerra, 2023-06-25:
Hi, I would like to use a memory on the apb bus.
How can I use this part of Murax Soc to use a ROM on the apb bus?
```
 val xip = ifGen(genXip)(new Area{
      val ctrl = Apb3SpiXdrMasterCtrl(xipConfig)
      ctrl.io.spi <> io.xip
      externalInterrupt setWhen(ctrl.io.interrupt)
      apbMapping += ctrl.io.apb     -> (0x1F000, 4 kB)

      val accessBus = new PipelinedMemoryBus(PipelinedMemoryBusConfig(24,32))
      mainBusMapping += accessBus -> (0xE0000000l, 16 MB)

      ctrl.io.xip.fromPipelinedMemoryBus() << accessBus
      val bootloader = Apb3Rom("src/main/c/murax/xipBootloader/crt.bin")
      apbMapping += bootloader.io.apb     -> (0x1E000, 4 kB)
    })
```

Thank you

MartinaBarreiroGuerra, 2023-06-26:
Yes, using onchip ram (using an fpga).

Dolu1990, 2023-06-26:
Ahh then i would say to not take example on the flash xip stuff, but instead take example on a regular peripheral, ex GPIO peripheral