SpinalHDL/VexRiscv #353: Error running MURAX on ARTY-A7 board and printing characters over Serial Port using println function

LongStudy, 2023-07-04:
I ran _MURAX_ on _ARTY-A7_ board according to the website below:
`https://github.com/SpinalHDL/VexRiscv/tree/master/scripts/Murax/arty_a7`
The LED does flash but the data from the **serial port is garbled**.

I don't quite know if the situation with this issue is the same with me:
`https://github.com/SpinalHDL/VexRiscv/issues/218`

First of all, I used the _ARTY-A7 100T_ board, except that the chip is different from 35t, the xdc constraint file should be consistent, so I modified `xc7a35ticsg324-1L` to `xc7a100tcsg324-1` in the .tcl file (and `xc7a35t` to `xc7a100t`).

Then I have modified the`/root/VexRiscv/SRC/main/c/murax/hello_world/SRC/main.c` and make it keep `println("hello world arty a7 v1");` in the loop. I tested the generated hex file `src/main/c/murax/hello_world/build/hello_world.hex` in `MuraxWithRamInit` and successfully loop output `hello world arty a7 v1`.

according to the website `https://github.com/SpinalHDL/VexRiscv/tree/master/scripts/Murax/arty_a7`, I ran `make` and `make prog/make flash` in sequence, and I did manage to compile and run. Then the LED does flash but the data from the **serial port is garbled**. My serial port parameters are 115200 and 8N1, the expected `hello world arty a7 v1` was changed to hexadecimal `94 94` or `17 F6` or `33 F2` with 2 Byte long.

As you mentioned in the `https://github.com/SpinalHDL/VexRiscv/issues/218`, I I suspect there is a shift in clock frequency. I want to know whether it is me or the script that causes the serial port to garble. If I am not the only one who has this problem would like to try to fix this error.

Dolu1990, 2023-07-10:
Thanks for the feedback :)
Where did you found the 115200 doc ?

LongStudy, 2023-07-11:
https://github.com/SpinalHDL/VexRiscv/tree/master/scripts/Murax/arty_a7#connect
The document was created earlier and the baud rate error may be caused by murax_arty resetting the clock rate to 100 Mhz but not resetting the serial port baud rate. I haven't looked at the serial divider in detail, so this reason is my guess.

Dolu1990, 2023-07-11:
Ahhhh XD
So, bydefault murax assume 12 Mhz clock and 115200 baudrate, and it happen that 100Mhz/12Mhz*115200=~1mbits/s XD (0.96 to be exact) 
So you would need to ajust the frequency setting in Murax

LongStudy, 2023-07-11:
That's what I thought, so I just set the baud rate to 1,000,000, and it work well.

closed 2023-07-11 by LongStudy