SpinalHDL/VexRiscv #354: How to add I/Dcache support to MURAX

LongStudy, 2023-07-08:
Hello, I want to know how to add **I/Dcache support to MURAX** for high performance , it seems that in addition to modifying IBusSimplePlugin and DBusSimplePlugin to their cache versions(IBusCachedPlugin/DBusCachedPlugin), also need to modify the bus link.

I see briey has config plugins for `Simple` and `Cached` I/DBUS.
```
for(plugin <- config.plugins) plugin match{
  case plugin : IBusSimplePlugin => iBus = plugin.iBus.toAxi4ReadOnly()
  case plugin : IBusCachedPlugin => iBus = plugin.iBus.toAxi4ReadOnly()
  case plugin : DBusSimplePlugin => dBus = plugin.dBus.toAxi4Shared()
  case plugin : DBusCachedPlugin => dBus = plugin.dBus.toAxi4Shared(true)
```

But murax only has config plugins for `Simple` I/DBUS.
```
for(plugin <- cpu.plugins) plugin match{
  case plugin : IBusSimplePlugin =>
    mainBusArbiter.io.iBus.cmd <> plugin.iBus.cmd
    mainBusArbiter.io.iBus.rsp <> plugin.iBus.rsp
  case plugin : DBusSimplePlugin => {
    if(!pipelineDBus)
      mainBusArbiter.io.dBus <> plugin.dBus
    else {
      mainBusArbiter.io.dBus.cmd << plugin.dBus.cmd.halfPipe()
      mainBusArbiter.io.dBus.rsp <> plugin.dBus.rsp
    }
  }
```

I don't know how to modify plugins to support IBusCachedPlugin/DBusCachedPlugin. If I only modify the I/DBusSimplePlugin of ArrayBuffer(cpuPlugins) to I/DBusCachedPlugin, sbt generates murax with the following error.
```
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.305 : Elaborate components
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: ??? vexriscv.MemoryTranslator
[error]         at scala.Predef$.assert(Predef.scala:170)
[error]         at spinal.core.package$.assert(core.scala:500)
[error]         at vexriscv.Pipeline$class.service(Pipeline.scala:26)
[error]         at vexriscv.VexRiscv.service(VexRiscv.scala:128)
[error]         at vexriscv.plugin.DBusCachedPlugin.setup(DBusCachedPlugin.scala:276)
[error]         at vexriscv.plugin.DBusCachedPlugin.setup(DBusCachedPlugin.scala:49)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error]         at vexriscv.VexRiscv.build(VexRiscv.scala:128)
[error]         at vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error]         at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:180)
[error]         at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:178)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.Component.prePop(Component.scala:178)
[error]         at spinal.core.Component.postInitCallback(Component.scala:187)
[error]         at vexriscv.demo.Murax$$anon$4.<init>(Murax.scala:240)
[error]         at vexriscv.demo.Murax.<init>(Murax.scala:227)
[error]         at vexriscv.demo.MuraxWithRamInit$$anonfun$main$6.apply(Murax.scala:546)
[error]         at vexriscv.demo.MuraxWithRamInit$$anonfun$main$6.apply(Murax.scala:546)
[error]         at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$66.apply(Phase.scala:2564)
[error]         at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error]         at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error]         at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 8 s, completed Jul 8, 2023 6:54:52 PM
```

Thank you for taking time out of your busy schedule to reply to our issues! Thanks a lot!

Dolu1990, 2023-07-10:
Hi,

So, the initial idea was that Murax is for cacheless design, while Briey is for cache full design.
There is currently no bridge for the i$ d$ to go to the main memory bus of Murax. Not that it would be necessarly something complicated to do, but Murax is mostly tuned to low latency cacheless stuff.

Would that be an issue to move on Briey ?

LongStudy, 2023-07-11:
Thanks a lot. I will try to run Briey on the arty board.

closed 2023-07-11 by LongStudy