SpinalHDL/VexRiscv #355: dev, sim: Error: fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01

likewise, 2023-07-16:
Currently I am developing against a local SpinalHDL `dev` branch, along with local VexRiscv `dev` and the VexRiscv-specific latest OpenOCD (i.e. using the "old" VexRiscv debug module") and I am seeing a possible regression during working with OpenOCD on my hardware. I am investigating. So I went back to VexRiscv/Sim/OpenOCD as per the README.

In simulation, I get different errors starting OpenOCD, but things do seem to work. But I do see errors I never saw before, are the following errors innocent?

This is the set of sources (current tips of `dev`, `dev`, riscv-spinal` branches, in this order) for:
```
git clone --branch 008c73f1ce18e294f137efe7a1442bd3f8fa2ee0 https://github.com/SpinalHDL/SpinalHDL.git
git clone --branch fd0f23abb686c1aa0c3bbcea5228d719992dca43 https://github.com/SpinalHDL/VexRiscv.git
git clone --branch 058dfa50d625893bee9fecf8d604141911fac125 https://github.com/SpinalHDL/openocd_riscv.git
```
```
/opt/openocd-vexriscv/bin/openocd --version
Open On-Chip Debugger 0.11.0+dev-04033-g058dfa50d (2023-07-16-15:23)
```

I am following the `README.md`:
```
#In the VexRiscv repository, to run the simulation on which one OpenOCD can connect itself =>
sbt "runMain vexriscv.demo.GenFull"
cd src/test/cpp/regression
make run DEBUG_PLUGIN_EXTERNAL=yes
```

```
vivado@vivado-container:/project-on-host/VexRiscv$ /opt/openocd-vexriscv/bin/openocd -c "set VEXRISCV_YAML /project-on-host/VexRiscv/cpu0.yaml" -f target/vexriscv_sim.cfg
Open On-Chip Debugger 0.11.0+dev-04033-g058dfa50d (2023-07-16-15:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/project-on-host/VexRiscv/cpu0.yaml
DEPRECATED! use 'adapter driver' not 'interface'
Info : only one transport option; autoselect 'jtag'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter srst delay' not 'adapter_nsrst_delay'
Info : set servers polling period to 50ms
Info : clock speed 4000 kHz
Info : TAP fpga_spinal.bridge does not have valid IDCODE (idcode=0x0)
Info : TAP auto0.tap does not have valid IDCODE (idcode=0x80000000)
Info : TAP auto1.tap does not have valid IDCODE (idcode=0xc0000000)
Info : TAP auto2.tap does not have valid IDCODE (idcode=0xe0000000)
Info : TAP auto3.tap does not have valid IDCODE (idcode=0xf0000000)
Info : TAP auto4.tap does not have valid IDCODE (idcode=0xf8000000)
Info : TAP auto5.tap does not have valid IDCODE (idcode=0xfc000000)
Info : TAP auto6.tap does not have valid IDCODE (idcode=0xfe000000)
Info : TAP auto7.tap does not have valid IDCODE (idcode=0xff000000)
Info : TAP auto8.tap does not have valid IDCODE (idcode=0xff800000)
Info : TAP auto9.tap does not have valid IDCODE (idcode=0xffc00000)
Info : TAP auto10.tap does not have valid IDCODE (idcode=0xffe00000)
Info : TAP auto11.tap does not have valid IDCODE (idcode=0xfff00000)
Info : TAP auto12.tap does not have valid IDCODE (idcode=0xfff80000)
Info : TAP auto13.tap does not have valid IDCODE (idcode=0xfffc0000)
Info : TAP auto14.tap does not have valid IDCODE (idcode=0xfffe0000)
Info : TAP auto15.tap does not have valid IDCODE (idcode=0xffff0000)
Info : TAP auto16.tap does not have valid IDCODE (idcode=0xffff8000)
Info : TAP auto17.tap does not have valid IDCODE (idcode=0xffffc000)
Info : TAP auto18.tap does not have valid IDCODE (idcode=0xffffe000)
Info : TAP auto19.tap does not have valid IDCODE (idcode=0xfffff000)
Warn : Unexpected idcode after end of chain: 21 0xfffff800        <----------------------------------
Error: double-check your JTAG setup (interface, speed, ...)
Error: Trying to use configured scan chain anyway...
Error: fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01       <----------------------------------
Warn : Bypassing JTAG setup events due to errors     <----------------------------------
[fpga_spinal.cpu0] Target successfully examined.
Info : starting gdb server for fpga_spinal.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
requesting target halt and executing a soft reset
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

```
/project-on-host$ /opt/x-tools/riscv32-unknown-elf/bin/riscv32-unknown-elf-gdb VexRiscv/src/test/resources/elf/uart.elf 
GNU gdb (crosstool-NG 1.25.0) 11.2
```
```
JTAG scan chain interrogation failed: all ones
Check JTAG interface, timings, target power, etc.
Trying to use configured scan chain anyway...
fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01    <----------------------------------
Bypassing JTAG setup events due to errors    <----------------------------------
(gdb) load
Loading section .yolo, size 0x1b8 lma 0x0
Loading section .text, size 0x160 lma 0x40000000
Start address 0x00000138, load size 792
Transfer rate: 6336 bits in <1 sec, 396 bytes/write.
(gdb) continue
Continuing.
(gdb)
```

Dolu1990, 2023-07-19:
the verilator testbench of VexRiscv alone doesn't implement JTAG, but some more direct connections.
So openocd get confused as it doesn't know about it.

You can ignore all the JTAG related error it throws at you ^^

likewise, 2023-07-19:
My origin problem is that `dev` branches show regressions on my hardware, i.e. I cannot debug my VexRiscv software anymore.

I was hoping I could replicate the problem in simulation.

How should I try to debug this regression in simulation?

Dolu1990, 2023-07-19:
Ahhh, then you can use the MuraxSim (SpinalSim), See : 
https://github.com/SpinalHDL/VexRiscv#murax-soc

This virtualize a proper jtag. Hardware issue should be reproducable there aswell.

What kind of regression exactly do you get on your hardware ?