SpinalHDL/VexRiscv #356: In configuration modified based on briey, the lw instruction cannot access the unaligned address

LongStudy, 2023-07-23:
Hi, I'm porting briey's architecture to murax to run on FPGA, I copied briey and commented the vga and sdram sections.

At first everything was fine, can running coremark etc., but when I compiled my own code, it stopped `lw`'s access to unaligned addresses and found the bug by printing.

```
   printf("addr = %d\n", *addr);
    asm  volatile(
    "mv     t1,%[add]          \n"       
    "lw     t3, 0(t1)       \n"
    .....
```

For example, in the code snippet above, the address prints out "33", and the simulation stops at the `lw` line. But if it print a number that multiple of four, it can continue to run.

It seems that the reason is that address is not aligned, that is, the address bits are not divisible by 4. I'm wondering if it's because of the AXI bus setting or the pipeline architecture.

LongStudy, 2023-07-23:
I briefly studied the code and found that the `catchUnaligned `Plugin option of `DBusCachedPlugin `seems to support non-aligned access if `catchUnaligned `is set to false. Is my understanding correct? In addition, will the setting have an impact on performance?

Dolu1990, 2023-07-23:
Hi,

VexRiscv data cache does not support unaligned access. 
catchUnaligned is false => assume there will never be a unaligned access (save area, good for configuration without exception handeling)
catchUnaligned is true => will trap on unaligned access (safe, but cost some are)

LongStudy, 2023-07-23:
Thanks for your prompt reply, I found some previous understanding errors, so if the hardware does not support unaligned access, do I need to deal with it on the software?

LongStudy, 2023-07-23:
I tried to fix it on the software using inline assembly, but unfortunately if the code elsewhere contains Pointers to 32-bit unaligned access addresses, the GCC-generated code will still enter a trap exception, and the number of code areas to troubleshoot will increase considerably. 

I wonder if there is any way to implement `lw` unaligned access in hardware (In addition to deleting Dcache).

Dolu1990, 2023-07-24:
> do I need to deal with it on the software?

Yes

On which software exactly do you ahve the missaligned pointer ?

> I wonder if there is any way to implement lw unaligned access in hardware (In addition to deleting Dcache).

For the cacheless VexRiscv, it should not be hard to add support for unaligned accesses, it is mostly just about handling the byte mask and shift.
Would you like to try it ?
Else, what is done in other cases, is to emulate unaligned access in software. When you get a unaligned access exception, you could run some software which would do the access using only 1 byte accesses.

That's how opensbi handle ii.

LongStudy, 2023-07-24:
You were right, I past thought the `DBusSimplePlugin` supported unaligned access, but when I tested it, the data was incorrect, and I found that the value of `catchAddressMisaligned` was false. This means that trap exception will not be entered, just like `catchUnaligned` for `DBusCachedPlugin`.

I've tried to ask a few people, and the unaligned access is uncomm. Because I've pointed the int32 pointer to an address that points to int8, so for an int8 address that's only 1 byte aligned, there's obviously a 3/4 chance it won't align to 4 bytes.

Considering that hardware misalignment may consume twice the alignment read time, whereas software may be able to increase the only 1 read time by concatenating back and forth, I would prioritize fixing it in software.

Thank you very much for your reply. In the future, I may try to modify the hardware design to achieve non-aligned access.