SpinalHDL/VexRiscv #357: Dcahce single burst on AXI write

hszilard13, 2023-07-25:
Hi,  I noticed  that the data cache is reading in bursts of 8, but when writing the data the `dBusAxi_aw_payload_len` signal is stuck at 0 and  `dBusAxi_a_payload_last` is always 1. 
[image]
The cache configuration is: 
```
 new DBusCachedPlugin(
          dBusCmdMasterPipe = true,
          dBusCmdSlavePipe = true,
          dBusRspSlavePipe = true,
          config = new DataCacheConfig(
            cacheSize         = 4096,
            bytePerLine       = 32,
            wayCount          = 1,
            addressWidth      = 32,
            cpuDataWidth      = 32,
            memDataWidth      = 32,
            catchAccessError  = true,
            catchIllegal      = true,
            catchUnaligned    = true,
            withLrSc = true,
            withAmo = true
          )
```
My expectation is if the cache line is 32 bytes wide that the CPU will write a whole line that is transfers of burst length 7.  Is this correct? Can I obtain maximum burst length?

dockside-code, 2023-07-25:
I think Vex only uses the bursting feature during read, in which it loads an entire cache line to the D$. AFAIK the write transfers are always issued in single beat.

Dolu1990, 2023-07-25:
VexRiscv has a write through cache.
So reads to fill the cache are always the size of a full cache line, while software writes directly goes to the main memory untouched. This behaviour can't be changed.

closed 2023-07-25 by hszilard13