SpinalHDL/VexRiscv #364: Unsupported compressed instruction on VexRiscvAxi4LinuxPlicClint.scala

PedroAntunes178, 2023-09-18:
Hi @Dolu1990 again,

I simply added the following to `VexRiscvAxi4LinuxPlicClint.scala`:
```
 compressedGen = true,
```
So that the CPU would support compressed instructions, and obtained the following error:
```
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.921 : Elaborate components
[info] PcManagerSimplePlugin is now useless
[error] Exception in thread "main" java.lang.Exception: Missing inserts : INSTRUCTION_ANTICIPATED
[error] 	at vexriscv.Pipeline$class.build(Pipeline.scala:95)
[error] 	at vexriscv.VexRiscv.build(VexRiscv.scala:126)
[error] 	at vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:180)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:178)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.Component.prePop(Component.scala:178)
[error] 	at spinal.core.Component.postInitCallback(Component.scala:187)
[error] 	at vexriscv.demo.VexRiscvAxi4LinuxPlicClint$$anonfun$1.apply(VexRiscvAxi4LinuxPlicClint.scala:106)
[error] 	at vexriscv.demo.VexRiscvAxi4LinuxPlicClint$$anonfun$1.apply(VexRiscvAxi4LinuxPlicClint.scala:18)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$67.apply(Phase.scala:2606)
[error] 	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 6 s, completed Sep 18, 2023, 11:07:55 PM
```

PedroAntunes178, 2023-09-18:
This happens on current master branch with sbt version:
```
sbt version in this project: 1.6.0
sbt script version: 1.8.2
```

PedroAntunes178, 2023-09-18:
Already noticed the comment in the documentation.
```
If you get a Missing inserts : INSTRUCTION_ANTICIPATE error, that's because the RegFilePlugin is configured to use SYNC memory read ports to access the register file, but the IBus plugin configuration can't provide the instruction's register file read address one cycle before the decode stage. To workaround that you can :

    Configure the RegFilePlugin to implement the register file read in a asyncronus manner (ASYNC), if your target device support such things
    If you use the IBusSimplePlugin, you need to enable the injectorStage configuration
    If you use the IBusCachedPlugin, you can either enable the injectorStage, or set twoCycleCache + twoCycleRam to false.

```

Thank you.

closed 2023-09-18 by PedroAntunes178

Dolu1990, 2023-09-19:
<3