SpinalHDL/VexRiscv #365: Where are the CLINT and PLIC described?

PedroAntunes178, 2023-09-18:
Hi,

I have a question rather than an issue. I'm wondering where the descriptions for CLINT and PLIC can be found. When I generate the VexRiscvAxi4LinuxPlicClint.v file, it seems that CLINT and PLIC are already included within the core. However, I couldn't locate the hardware descriptions for CLINT and PLIC in the VexRiscv repository.

Thank you in advance.

Dolu1990, 2023-09-19:
Personnaly, i reversed engineered them from various software drivers out there in the wild XD
Bit a paine in the ass.

They are in the SpinalHDL repository

Dolu1990, 2023-09-19:
https://github.com/SpinalHDL/SpinalHDL/blob/645e03f21326f9baffe49ac4c59cff2ea6e86cd3/lib/src/main/scala/spinal/lib/misc/plic/PlicMapper.scala#L37
https://github.com/SpinalHDL/SpinalHDL/blob/645e03f21326f9baffe49ac4c59cff2ea6e86cd3/lib/src/main/scala/spinal/lib/misc/Clint.scala#L49

Dolu1990, 2023-09-19:
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/plic.h
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/clint.h

PedroAntunes178, 2023-10-10:
Hi,
Are we ablle to set the MTIME FREQ, in the CLINT?
From looking at the [Scala](https://github.com/SpinalHDL/SpinalHDL/blob/645e03f21326f9baffe49ac4c59cff2ea6e86cd3/lib/src/main/scala/spinal/lib/misc/Clint.scala), it seems the MTIMER counter [increments every clock cycle](https://github.com/SpinalHDL/SpinalHDL/blob/645e03f21326f9baffe49ac4c59cff2ea6e86cd3/lib/src/main/scala/spinal/lib/misc/Clint.scala#L39). Would it be possible to define in the VexRiscv Scala file that the internal timer frequency is 100 KHz? Therefore, the MTIMER counter would only increment every `CLK_FREQ / MTIME_FREQ` clock cycles.

Dolu1990, 2023-10-11:
Hi,

would need to modify the clint implemention, would not be complicated