SpinalHDL/VexRiscv #367: sbt fail: [error] error while loading <root>, Error accessing... scalactic_2.11-3.2.5.jar

pottendo, 2023-09-29:
Hi,
sorry upfront, as I assume this is a local build-env issue, but it's driving me nuts... 
I'm working on [this](https://github.com/pottendo/RVCop64-pottendo/tree/wip-mailbox) project.

As soon as sbt is involved to rebuild certain elements within VexRiscV it fails here:
```
...
[info] compiling 351 Scala sources to /opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/lib/target/scala-2.11/classes ...
[error] error while loading <root>, Error accessing /home/pottendo/.cache/coursier/v1/https/repo1.maven.org/maven2/org/scalactic/scalactic_2.11/3.2.5/scalactic_2.11-3.2.5.jar
[error] ## Exception when compiling 351 sources to /opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/lib/target/scala-2.11/classes
[error] scala.reflect.internal.MissingRequirementError: object java.lang.Object in compiler mirror not found.
...
```
this used to work for some time, but probably after some upgrade of my dev machine it fails now.

the command invoked is:
```
cd /opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv && \
sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=64 --dbus-width=64 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=32 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=True --fpu=True --cpu-per-fpu=4 --rvc=True --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw64Is4096Iy1_Dw64Ds4096Dy1_ITs4DTs4_Ood_Wm_Fpu4_Rvc --netlist-directory=/opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4"
```
I can confirm that 
```
/home/pottendo/.cache/coursier/v1/https/repo1.maven.org/maven2/org/scalactic/scalactic_2.11/3.2.5/scalactic_2.11-3.2.5.jar
```
exists and is readable.

I use:
```
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
$ dpkg -l |grep sbt
ii  sbt                                        1.9.6                                      all          sbt, the interactive build tool
$ sbt --version
[info] [launcher] getting org.scala-sbt sbt 1.9.6  (this may take some time)...
[info] [launcher] getting Scala 2.12.18 (for sbt)...
sbt version in this project: 1.9.6
sbt script version: 1.9.6
```
I followed the instruction to get sbt from its repos.

Any hint or idea?
thx, pottendo

Dolu1990, 2023-09-29:
Yes, that's quite enoying that things break because of some cloud related stuff XD

Soluton is to add in the build.sbt : 
libraryDependencies += "org.scalactic" %% "scalactic" % "3.2.10"

Did you updated the pythondata-cpu-vexriscv_smp submodules ?
I think it was fixed not that long ago

pottendo, 2023-09-29:
...as always, one can count on @Dolu1990! Thanks - this fixes the build here.
Unfortunately I can't update to the master trees and its submodules, as this breaks the build entirely. It seems that then some other modules (litex?) require updates to the most recent versions, which ultimately breaks the project mentioned above.
I need to sync with the original developer (@zeldin) whether an upgrade of all required litex modules is feasible for him - this may potentially enable the addon of my vexriscv_smp fork for this project.

Unfortunately this is all beyond my knowledge to do it myself.

anyway, thanks for helping so quick!
pottendo

closed 2023-09-29 by pottendo

Dolu1990, 2023-09-29:
Adding this line should fix it with any version : 
https://github.com/SpinalHDL/SpinalHDL/blob/8007a68fd6194e70a9b233cc24cbc3d8bea6a835/build.sbt#L30

(in pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/build.sbt)

Good luck ^^

pottendo, 2023-09-29:
> Adding this line should fix it with any version : https://github.com/SpinalHDL/SpinalHDL/blob/8007a68fd6194e70a9b233cc24cbc3d8bea6a835/build.sbt#L30
> 
> (in pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/build.sbt)
> 
> Good luck ^^

yes, I did add a similar already - works, as expected. Need to cherry pick some more from the master tree - the FPU context switch stuff wasn't in my baseline and I need to get those merged into this again.

thanks for responding and in general for these project - it's great to see what open source can achieve!
bye, pottendo