SpinalHDL/VexRiscv #368: Read speed: dcache width impact 32bit vs 64bit

pottendo, 2023-09-29:
Hello,

again with reference to [this](https://github.com/pottendo/RVCop64-pottendo/tree/wip-mailbox) project I observed the following substantial performance difference when having different widths of `--dcache-width`:

```
################# 32 bit default

$ sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=32 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=True --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm --netlist-directory=/opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4"

$ litex_term /dev/ttyACM0
[...]
--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 80MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		48KiB
SRAM:		16KiB
MAIN-RAM:	16384KiB 

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.8MiB/s
   Read speed: 61.0MiB/s

--============== Boot ==================--
Booting from serial...

################# 64 bit --dcache-width=64

$ sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=32 --dbus-width=64 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=32 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=True --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw64Ds4096Dy1_ITs4DTs4_Ood_Wm --netlist-directory=/opt/work/src/orangeCart/RVCop64-scratch/RVCop64-pottendo/hw/deps/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4"

 [...]

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 80MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		48KiB
SRAM:		16KiB
MAIN-RAM:	16384KiB 

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.8MiB/s
   Read speed: 13.3MiB/s

--============== Boot ==================--
Booting from serial...
```

as one can see the read speed differs substantially - which of course results in lesser overall performance with 64bit dcache width on this architecture / board.

I wouldn't bother but enabling `--with-fpu` or `--cpu-count=2` implicitly sets 64bit dcache width - so the overall performance gain due to a FPU or even dual core CPU is limited due to the overall slower memory read speed. This is unfortunate, as FPU or multithreaded applications may not be the majority, but handy from time to time...

Anything what one can do about it?

thx, pottendo

PS: Some Caveat: 
I couldn't run these tests on the most recent VexRisxVSMP CPUs (see [here](https://github.com/SpinalHDL/VexRiscv/issues/367)). 
In case some enhancements have been done in the last 8-10 months affecting this area, this may be 'fixed'... 
Thanks for understanding.

Dolu1990, 2023-09-29:
Hi,

Ahhh that's weird that the performance drop.
I would say, the only way to figure that one out would be to run it in simulation to see what is happening exactly.

Seems to me that Read speed: 61.0MiB/s is the outliner XD, as VexRiscv normaly get a higher write speed.

pottendo, 2023-09-29:
hmm,
too long since I've done something with the simulator... You've helped to track down some FPU / multi-threaded [issue](https://github.com/SpinalHDL/VexRiscv/issues/297) on Zephyr/Linux, if you remember.
Can you point me to the right resources, where I can study, how to run this particular bitstream on an emulator? or whatever piece of it... 
Or do I need to extract the memtest to a standalone (e.g. Zephyr) program?
bye, pottendo

pottendo, 2023-09-30:
hi,
I've run naively:

64bit:
```
$ litex_sim --sim-debug --cpu-type vexriscv_smp --dcache-width=64 --with-sdram
[...]
--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		64.0MiB 32-bit @ 1MT/s (CL-2 CWL-2)
MAIN-RAM:	64.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (8.0KiB)...
  Write: 0x40000000-0x40002000 8.0KiB   
   Read: 0x40000000-0x40002000 8.0KiB   
Memtest OK
Memspeed at 0x40000000 (Sequential, 8.0KiB)...
  Write speed: 1.6MiB/s
   Read speed: 871.5KiB/s

--============== Boot ==================--

litex> mem_speed 0x40000000 102400
Memspeed at 0x40000000 (Sequential, 100.0KiB)...
  Write speed: 1.6MiB/s
   Read speed: 878.7KiB/s
```

and here the 32bit:

```
$ litex_sim --sim-debug --cpu-type vexriscv_smp --dcache-width=32 --with-sdram
--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		64.0MiB 32-bit @ 1MT/s (CL-2 CWL-2)
MAIN-RAM:	64.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (8.0KiB)...
  Write: 0x40000000-0x40002000 8.0KiB   
   Read: 0x40000000-0x40002000 8.0KiB   
Memtest OK
Memspeed at 0x40000000 (Sequential, 8.0KiB)...
  Write speed: 1.6MiB/s
   Read speed: 874.6KiB/s
[...]
--============= Console ================--

litex> mem_speed 0x40000000 102400
Memspeed at 0x40000000 (Sequential, 100.0KiB)...
  Write speed: 1.6MiB/s
   Read speed: 878.8KiB/s
```

No difference, 64bit vs. 32bit dcache-width, and confirmed that the write speed is faster.

Not sure if this experiment can even be representative for my observation on the real hardware.
What I can confirm is, that the performance on the real FPGA HW is substantially different. I've done a testcase where the one with `--dcache-width=64` takes ~625s and with `--dcache-width=32` its around 125s and even fast with `--cpu-type=vexriscv` (32bit dcache default) it's around 114s.

bye, pottendo

Dolu1990, 2023-09-30:
Hi ^^

What memory system do you use on hardware ? (DDR, ... ) ?

Regards
Charles

pottendo, 2023-09-30:
Well, don't know more than it's 'Hyperram'.
The partlist is here: https://github.com/zeldin/OrangeCart
My project is a fork of the RVCop64 project.

Can't open kicad files, so I can't tell you more, Sorry.

Bye, pottendo

Dolu1990, 2023-09-30:
Ahhh then, one possibility, is that the hyperram controller successfuly reconize 32 bits based burst, but fail at reconizing the 64 bits based ones ?

pottendo, 2023-09-30:
...must be something like this, I suppose.
Here some more information: https://github.com/zeldin/litehyperram/tree/22123da560144fdc7113a1d72f1bda8e4beea5da
At least some information about bursts: `- Arbitrary burst length` ...

pottendo, 2023-09-30:
oops, mistakenly closed -> however, this is probably off-topic here, so @Dolu1990, it's ok to be closed.
Thanks for the feedback so far - maybe there's a solution; need to check with @zeldin.

pottendo, 2023-10-13:
@zeldin has fixed the issue - thanks @Dolu1990 and @zeldin!
as always, you guys rock!

closed 2023-10-13 by pottendo

Dolu1990, 2023-10-13:
<3