SpinalHDL/VexRiscv #370: Interface BSCAN2 with VexRiscv's JTAG

ztachip, 2023-10-09:
Hi
I am trying to use Xilinx BSCAN2 for JTAG debugging but I try to instantiate BSCAN2 outside of VexRiscv and just route the JTAG signals to VexRiscv (based on Briey.scala design with JTAG).
The reason is that I try to keep my project easy to port between different FPGA by isolating the portable HDL codes from the components that are specific to FPGA silicon. That's why I look for way to do JTAG over BSAN2 but without referencing VexRiscv's jtagCtrl.fromXilinxBscane2 function since I would like to keep the soft-CPU to be the portable component.
But so far openocd fails to communicate with VexRiscv this way.
Should this work or there are some logic to apply to BSAN2's signals before they can be routed to VexRiscv.
Thanks

Dolu1990, 2023-10-11:
Hi

Have you seen this doc ? :
https://github.com/SpinalHDL/VexRiscv/tree/master/doc/nativeJtag
Which bscane2 id did you used ?

ztachip, 2023-10-11:
Hi
I am working on a project that I try to keep it portable between different FPGA as much as possible.
Yes, I did try the method described in the doc and it works well.
But what I try to do is to instantiate BSCAN2 outside of VexRiscv.
This way VexRiscv remains portable between different FPGA since it will just reference standard JTAG signal like TCLK,TDO,TDI,,,
Wondering if you can share information how you convert BSCAN2 to JTAG signals.
Thanks

https://github.com/SpinalHDL/VexRiscv/tree/master/doc/nativeJtag

Dolu1990, 2023-10-13:
Ahh i would say you would have two ways,
Either you expose the jtag instruction interface (bscane like) which can be then bridged to something else. And in the case you want to bridge it from a true 4 pins jtag, you just need to create a little jtag tap which "behave like the bscane2"

Something like : 

```scala
       val jtag = slave(new Jtag())
        val jtagClockDomain = ClockDomain(jtag.tck) on new Area
          val tap = new JtagTap(jtag.io, 4)
          val idcodeArea = tap.idcode(B(p.tapId, 32 bits))(5)
          val wrapper = tap.map(debug.jtagInstruction, instructionId = 8)
        }
        
```        

Either you expose the native vex debug interface, which is a small memory bus, and then you provide the bridge from what ever -> to it.

closed 2023-10-22 by ztachip