SpinalHDL/VexRiscv #372: Adding FPU to AHBLite3 config file

vidushiy25, 2023-10-18:
Hi, I wanted to add the FPU plugin to the AHBlite3 config file so I can convert everything to Verilog but doing this is giving me errors. I have attached the modified file. Do let me know what's wrong.
[VexRiscvAhbLite3.scala.zip](https://github.com/SpinalHDL/VexRiscv/files/13030328/VexRiscvAhbLite3.scala.zip)

vidushiy25, 2023-10-18:
Hi, I tried doing it and it gives more errors when i use the command :sbt "runMain vexriscv.demo.VexRiscvAhbLite3"

[info] welcome to sbt 1.6.0 (Oracle Corporation Java 19.0.2)
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /Users/vidushiyaksh/Desktop/VexRiscv/project
[info] loading settings for project root from build.sbt ...
[info] set current project to VexRiscv (in build file:/Users/vidushiyaksh/Desktop/VexRiscv/)
[info] running (fork) vexriscv.demo.VexRiscvAhbLite3 
[info] [Runtime] SpinalHDL v1.9.3    git head : 029104c77a54c53f1edda327a3bea333f7d65fd9
[info] [Runtime] JVM max memory : 4096.0MiB
[info] [Runtime] Current date : 2023.10.18 16:32:41
[info] [Progress] at 0.000 : Elaborate components
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.231 : Elaborate components
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: ??? vexriscv.plugin.DBusEncodingService
[error] 	at scala.Predef$.assert(Predef.scala:170)
[error] 	at spinal.core.package$.assert(core.scala:500)
[error] 	at vexriscv.Pipeline$class.service(Pipeline.scala:26)
[error] 	at vexriscv.VexRiscv.service(VexRiscv.scala:126)
[error] 	at vexriscv.plugin.FpuPlugin.setup(FpuPlugin.scala:175)
[error] 	at vexriscv.plugin.FpuPlugin.setup(FpuPlugin.scala:25)
[error] 	at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error] 	at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error] 	at vexriscv.VexRiscv.build(VexRiscv.scala:126)
[error] 	at vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:180)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:178)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.Component.prePop(Component.scala:178)
[error] 	at spinal.core.Component.postInitCallback(Component.scala:187)
[error] 	at vexriscv.demo.VexRiscvAhbLite3$$anonfun$1.apply(VexRiscvAhbLite3.scala:145)
[error] 	at vexriscv.demo.VexRiscvAhbLite3$$anonfun$1.apply(VexRiscvAhbLite3.scala:25)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$67.apply(Phase.scala:2606)
[error] 	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1

The file with the import line : 
[AHBLITE3FPU.zip](https://github.com/SpinalHDL/VexRiscv/files/13030621/AHBLITE3FPU.zip)

Dolu1990, 2023-10-18:
Ahhh that's because the FPU only support the data cache, so you need to use DBusCachedPlugin instead of DBusSimplePlugin

Dolu1990, 2023-10-18:
uncomment the 2 lines : 
//            memoryTranslatorPortConfig = null
...
//          ),

Dolu1990, 2023-10-18:
you can remove : 
,
             catchMemoryTranslationMiss = true

vidushiy25, 2023-10-18:
Thanks, it works. How would I connect this to an AHB interconnect? Would I just need to connect the iBusAhbLite3 signals to the AHB interconnect or do I also need to figure out how to connect the dBus signals?

closed 2023-10-18 by vidushiy25