SpinalHDL/VexRiscv #373: Scratchpad memory with cached IBUS and DBUS?

piondeno, 2023-10-30:
Dear all
Could IBUS and DBUS of Vexriscv have scratchpad memory (1T) for high speed operation?
At the same time, it also provide cached IBUS and DBUS for low speed memory (Flash memory).
Is it possibile to config Vexriscv for above requirements?

Dolu1990, 2023-10-30:
Hi, what do you mean by 1T ?

Currently, thigly coupled ram is supported for i$. for the d$ the user can use a peripheral memory mapped in IO region, but at a performance cost

piondeno, 2023-10-31:
Hi, Dolu1990
Thanks for your reply.
The 1T what I mean is that can IBusCachedPlugin  fetch instruction only 1 clock delay after sending vailde PC address in some memory region.

Tightly coupled ram is that I want and I will try to generate TCM for testing it.
Thanks a lot.

piondeno, 2023-10-31:
Hi, Dolu1990
Thanks for reply again.
I am trying to implement TCM. 
But I am a new one in SpinalHdl, encounter errors continuously.
I try to find the reference code for TCM and seems like none in "demo" folder.

I write the following code to generate TCM, but not success.
      for(plugin <- config.plugins) plugin match{
        //case plugin : IBusSimplePlugin => iBus = plugin.iBus.toAxi4ReadOnly()
        case plugin : IBusCachedPlugin => {
          iBus = plugin.iBus.toAxi4ReadOnly()
          val iBusTc=plugin.newTightlyCoupledPortV2(TightlyCoupledPortParameter("iBusTc", a => a(31 downto 28) === 0x0 && a(10)))
          val tcmMemDataWidth = log2Up(1024)
          val tcmMem = new Mem(Bits(32 bits), wordCount = 1024)
          iBusTc.data := tcmMem.readSync(iBusTc.address((tcmMemDataWidth-1) downto 0),iBusTc.enable)
        }

Could you provide me a simple demo for TCM?
Thanks a lot.

Dolu1990, 2023-10-31:
Hi, I will add a demo

Dolu1990, 2023-10-31:
Done, see https://github.com/SpinalHDL/VexRiscv/blob/281818af9c04517baa7c6dffe62e229bfa7e036a/src/main/scala/vexriscv/demo/GenFullWithTcm.scala

piondeno, 2023-11-01:
Thanks for the demo.
After download the file and run this demo, it shows one error.

`/home/datakey/tools/VexRiscV/VexRiscv-Idea/src/main/scala/vexriscv/demo/GenFullWithTcm.scala:55:40
type mismatch;
 found   : vexriscv.plugin.TightlyCoupledDataPortParameter
 required: spinal.core.UInt => spinal.core.Bool
        TightlyCoupledDataPortParameter("dBusTc", a => a(31 downto 28) === 0x3)`

Do you encounter same status?
If it only occurs on my project, I will re-clone whole project again.

I mark line 54 and 55 off and only generate TCM for IBusCachedPlugin.
It can generate verilog code successfully.

I plan to use a single-port ram to implement TCM and make sure debug-port also can assess this TCM.
Thanks a lot.

Dolu1990, 2023-11-02:
Hi,
> After download the file and run this demo, it shows one error.

You may have a too hold vexriscv, need to update it i guess.

Also, note that i pushed another example which integrate the memory directly in VexRiscv, but as a dual ported blackbox : 
https://github.com/SpinalHDL/VexRiscv/commit/beeec94344bd0caede9b0d91aee3e198d8b6f5b3#diff-c6e62ed4d29d36f377790574498f2dc058f029a2a88e0c5b45f22d502658aaca

piondeno, 2023-11-16:
Hi Dolu1990, thanks for your help.

After re-download the vexriscv,  there is no error on previous demo file.
I will try the new demo file, thanks a lot.

piondeno, 2023-11-20:
Hi @Dolu1990,

May I ask for your help?

First, I want to single-port to implement TCM and rewrite the code for build.
```
  override def build(pipeline: VexRiscv) = {
    val logic = pipeline plug new Area {
      val ram = Mem(Bits(32 bits), mapping.size.toInt/4)
//      ram.generateAsBlackBox()
      val address = Mux(ibus.enable,(ibus.address >> 2),(dbus.address >> 2)).resized
      val tcmRam = new Area{
        val readData = ram.readWriteSync(
            address = address,
            data    = dbus.write_data,
            enable  = (dbus.enable | ibus.enable),
            write   = dbus.write_enable,
            mask    = dbus.write_mask
          )
        dbus.read_data := readData
        ibus.data := readData
      }
```

Second, I generate two TCM for ITCM and DTCM.
```
  val ITCM = new IBusDBusCachedTightlyCoupledRam(
    mapping = SizeMapping(0x20000000, 0x2000)
  )
  val DTCM = new IBusDBusCachedTightlyCoupledRam(
    mapping = SizeMapping(0x30000000, 0x1000)
  )
```

Third, modify the ld file to seperate code and data. 
And seperate the hex file as two parts for simulation.
One is for ITCM and the other for DTCM.
```
MEMORY {
  iTcm (WX!R)/*(RX)*/ : ORIGIN = 0x20000000, LENGTH = 8K
  dTcm (W!RX) : ORIGIN = 0x30000000, LENGTH = 4K
}
_stack_size = 2k;
_heap_size = 0k;

SECTIONS /*TODO don't initialize useless things,  restore literal loading that use 2 instruction in place of onChipRam word */
{
  .vector : {
    *crt.o(.text);
  } > iTcm

  .memory : {
    *(.text);
    end = .;
  } > iTcm

  .rodata         :
  {
    *(.rdata)
    *(.rodata .rodata.*)
    *(.gnu.linkonce.r.*)
  } > dTcm
```

Fourth, I try to use HexTools.initRam() to initial the content of TCM RAM. 
But I can not file the hierarchy than can point to the RAM of TCM.
Could you give me some suggestion ?
```
HexTools.initRam(toplevel.axi.???, "/home/datakey/myProject/VexRiscvSocSoftware-master/projects/briey/timer/build/timerITCM.hex", 0x20000000l)
HexTools.initRam(toplevel.axi.???, "/home/datakey/myProject/VexRiscvSocSoftware-master/projects/briey/timer/build/timerDTCM.hex", 0x30000000l)

```
[Ryan.zip](https://github.com/SpinalHDL/VexRiscv/files/13406100/Ryan.zip)

Dolu1990, 2023-11-25:
Hi,
I pushed more arguements to ramOffset
Currently, the idea of the IBusDBusCachedTightlyCoupledRam.
See https://github.com/SpinalHDL/VexRiscv/commit/7c6c7a6fe5aed7f55ae7119b95e5cd569f0efc90#diff-106774d0f914a0abb624dcdc0ad496761640bbabb69f5d287f5639e38debdb9aR654

So, you need to set ramAsBlackbox=false hexInit=pathOfHexFile ramOffset=offsetAtWhichTheRamIsMapped

piondeno, 2023-12-05:
Thanks for @Dolu1990  once again.

I still make some modification for using single port  sram like this

```scala
  override def build(pipeline: VexRiscv) = {
    val logic = pipeline plug new Area {
      val muxAddress = Mux(ibus.enable,(ibus.address >> 2),(dbus.address >> 2)).resized
      val muxEnable = (dbus.enable | ibus.enable)
      val ram = Mem(Bits(32 bits), mapping.size.toInt/4)
      if(ramAsBlackbox) ram.generateAsBlackBox()
      if (hexInit != null) {
        assert(ramOffset != -1)
        val temp = ramOffset.toBigInt
        initRam(ram, hexInit, temp, allowOverflow = true)
      }
      val tcmRam = new Area {
        val readData = ram.readWriteSync(
          address = muxAddress,
          data = dbus.write_data,
          enable = muxEnable,
          write = dbus.write_enable && dbus.enable,
          mask = dbus.write_mask
        )
        dbus.read_data := readData
        ibus.data := readData
      }
    }
  }
``` 
Now I can use load hex file to ITCM and DTCM directly and run the simulation successfully.
After programming the FPGA, I also can use debug port (JTAG)  to download program into ITCM and DTCM.
Thanks

closed 2023-12-05 by piondeno

Dolu1990, 2023-12-11:
Hi,

I'm just a bit scared with that modification, because it seems that if the ibus and dbus acess at the same time, then dbus request will be ignored ?

piondeno, 2023-12-12:
Hi,
[linkerTcm.zip](https://github.com/SpinalHDL/VexRiscv/files/13642539/linkerTcm.zip)

First, I use linkerTcm.ld to separate .text and .data 
And generate ITCM for .text and DTCM for .data
In the simulation, I did not see any request from dbus.enable in ITCM bus.

Second, run the debug mode in eclipse IDE.
It is OK to download program.
I can set break point , run single step and not encounter any problem and crash right now.
It seems like no collision in debug mode.

If you need more testing, please let me know.
Thanks

Dolu1990, 2023-12-18:
Ahhh ok ^^
I understand now,
Thanks