SpinalHDL/VexRiscv #389: DE0-Nano Board with VexRiscV: IO and Fit Design Issues Including Specific Command Used

Tahamermer, 2024-02-04:
Hello VexRiscV Community,

I've been trying to run my VexRiscV project on my DE0-Nano FPGA board and encountered two issues. I aimed to configure the project for a Cyclone IV FPGA following the guide on [VexRiscV GitHub](https://github.com/SpinalHDL/VexRiscv). However, when compiling my design in Quartus, I received the following errors:

Error (171000): Can't fit design in device - indicating that my design does not fit into the FPGA device I'm attempting to use.
Error (169282): There are 145 IO output pads in the design, but only 139 IO output pad locations available on the device. - indicating that my design has more IO output pads than what's available on the device.
For generating the Verilog code, I used the specific command sbt "runMain vexriscv.demo.GenFull". Is there a particular configuration or step required to adapt the design for the Cyclone IV? Do you have any suggestions or experiences in overcoming these errors?

Any help or guidance would be greatly appreciated. Thank you in advance!

Dolu1990, 2024-02-05:
Hi,

You need to select a FPGA with more IO.

Tahamermer, 2024-02-08:
Hello ,

Actually I want a add  some custom ISA functions in VexRiscv.Do you any advice ? 

I will try Briey SOC for de0 nano  
https://drive.google.com/drive/folders/0B-CqLXDTaMbKZGdJZlZ5THAxRTQ?usp=sharing
in this link. But I want to know about how can I try to integrate own custom instruction's in BrieySOC.

Any help or guidance would be greatly appreciated. Thank you in advance!

Dolu1990, 2024-02-12:
Hi,

I would say, add an instance of that plugin (https://github.com/SpinalHDL/VexRiscv?tab=readme-ov-file#add-a-custom-instruction-to-the-cpu-via-the-plugin-system)

into that list : 

https://github.com/SpinalHDL/VexRiscv/blob/fe5e6dd95b932ac16df1c306c5e6b5f223506e2b/src/main/scala/vexriscv/demo/Briey.scala#L39

That's the symplest way.