SpinalHDL/VexRiscv #393: EU Funding

PythonLinks, 2024-03-01:
The EU  has €270 million in funds for Risc-V projects. 

https://www.hpcwire.com/2022/12/16/europe-to-dish-out-e270-million-to-build-risc-v-hardware-and-software/

We should figure out how to get some of that money.  Presumably people here are already working on that.  How can I help?

Two obvious application areas are building a cpu with additional instructions, and chiplets, connecting up large numbers of risc-v cores in different ways. 

Of course where the real money is, is some cross between AI, and Risc-V, although personally I am not a fan of the AI stuff. Most inefficient. Energy consumption reportedly doubles every 4 months.  Very bad for climate change.

Dolu1990, 2024-03-01:
Hi,

I wasn't aware of it XD
So, i'm working on https://github.com/SpinalHDL/VexiiRiscv which aim to be a replacement for VexRiscv, would be nice to have some help on that ^^

My guess so far, is that 270M mean big compagny / big industry project / Big achademia. Probably not aiming at individual people doing open source :/

Or is it ?

PythonLinks, 2024-03-02:
Well let us find out if the EU is open to funding us or not. In general open source hardware, specifically Risc-v is a hot item with the EU government.   In particular SpinalHDL is great software, but  with (forgive me) terrible marketing.  The gorgeous Risc-V documentation lacks an introductory sections called "Why use VexRiscV".   It also lacks a section comparing it to the alternatives.   I suspect that even the section mentioning users is a bit weak.  I think it is missing from the index. 

It is hard to express the advantages of this approach. The mass market just wants a Verilog solution.  We need a strong elevator pitch.   Reading the VexRiscV intro and the SpinalHDL intro, I am not sold.  The elevator pitch is weak.  I am here because of my software background, because I understand how much better this approach is, not because you did a good job selling me on this approach.  

In particular we need a good description of why this is better than the Python based approaches.  Amaranth for example.  They even have a very mature TCP/IP stack. 

So let me try to pitch SpinalHDL. 

Open source hardware, written in Verilog, makes it possible to release a commercial Risc-v chip, VexRiscV makes it easy.   For all of those companies who want to release a Risc-V chip with custom instructions, SpinalHDL is the least expensive and fastest starting point.    It is not just because it leverages the power of Scala in the verification process.   It is not just because SpinalHDL is the only open source tool supporting assertions.  It is because SpinalHDL supports software abstractions. 

Verilog is great at hardware abstraction, building a hierarchy of circuits out of smaller components, but it completely lacks software abstraction.  You want to connect to a signal in a different part of the hierarchy, you have to  manually run wires up and down the hierarchy.  In SpinalHDL, you just name the signal you want to connect with.  

I will leave it to others, more experienced than me, in writing some more examples.    There are a bunch in the documentation.  

Anyhow, I believe that the exercise of finding out what the EU is funding, what the companies are seeking, what the VC's want, what the Silicon Velley vc.s want, is critical to our growth. 

What am I doing here?  I am the guy working on stack machines for my master's thesis. Think J1Sc.   Very fast context switches. 2nd semester is over, I got perfect scores in Verilog and veification, now my focus is on the thesis.  I have reached the limits of what verilog can do.  In verilog simulation, I made a change to the Forth code, and suddenly everything broke.  There is no error message.  Trial and error is way too slow an approach to debugging.  I need assertions. lots of assertions to tell me what went wrong.   When I do software debugging, really I am acting as a human assertion checker, with the J1 family, there are so many things going on at the same time, that assertion checking needs to be automated.  SpinalHDL gives me that.  I think it is the only open source tool that does it.    So SpinalHDL is what I need.  Of course the money is in the Risc-V world.  I like money.  Basically the tools are all the same.  I can do both.  In any case, it is important to understand the Risc-V market in order to understand the market niche for stack machines.

Dolu1990, 2024-03-04:
Hi @PythonLinks 

> but with (forgive me) terrible marketing

I had quite some energy for this 4-5 years ago.

> It is hard to express the advantages of this approach. 

That's very difficult, because most hardware engineer do not have any software background past low level C.
So, you can tell them what every story you want, they won't "connect".
The only way i found so far is to have a very sexy project to introduce people the idea that it work well (VexRiscv)

> Open source hardware, written in Verilog, makes it possible to release a commercial Risc-v chip, VexRiscV makes it easy. For all of those companies who want to release a Risc-V chip with custom instructions, SpinalHDL is the least expensive and fastest starting point.

From a compagny perspective, the hard point is to find poeple with the technical skills, So fondamentaly, SpinalHDL adoption is hard for large scale entities, realy have to focus on small groups. And so because of that ASIC focus isn't great (large scale stuff), have to focus on FPGA people, which are much smalller scale in general => a few motivated people can pickup SpinalHDL

I don't know, overall the situation is hard.
So far, my plan is to have VexiiRiscv (Vex2Risc5) as a "sexy" SpinalHDL project, then maybe work toward SoC stuff.
VexiiRiscv seems promessing.

In other words, "advertising" from the results.