SpinalHDL/VexRiscv #395: Compile C code and run bare metal cycle accurate simulation

nachiket, 2024-03-06:
Apologies for the naive question, but how do I run a cycle-accurate Verilator simulation of a simple C program, run it on the VexRisc processor RTL and calculate how many cycles it took to execute and also check functional correctness of the result? 

I'm running Ubuntu 18.x. I'm able to build a "full" version of the core and run the IndividualTests.  I also built the risc-v gnu toolchain. 

I see the RUN_HEX option in a regression test makefile, but I'm a bit unsure how to use it with my own C/C++ example code.

Thanks for your attention.

Dolu1990, 2024-03-08:
Hi,

Here is an example : 
https://github.com/SpinalHDL/VexRiscvSocSoftware/tree/master/projects/raw/simple

It is a bit dated XD
but it should work with the RUN_HEX option :)

nachiket, 2024-03-14:
I get this error if I type make in that folder:

```
mkdir -p build
mkdir -p build/src/
/opt/riscv//bin/riscv64-unknown-elf-gcc -c -march=rv32im  -mabi=ilp32 -g -O3  -MD -fstrict-volatile-bitfields    -o build/src/main.o src/main.c
mkdir -p build/src/
/opt/riscv//bin/riscv64-unknown-elf-gcc -c -march=rv32im  -mabi=ilp32 -g -O3  -MD -fstrict-volatile-bitfields  -o build/src/crt.o src/crt.S -D__ASSEMBLY__=1
src/crt.S: Assembler messages:
src/crt.S:90: Error: unrecognized opcode `csrw mie,a0', extension `zicsr' required
src/crt.S:92: Error: unrecognized opcode `csrw mstatus,a0', extension `zicsr' required
../../../resources/subproject.mk:70: recipe for target 'build/src/crt.o' failed
make: *** [build/src/crt.o] Error 1
```

Dolu1990, 2024-03-15:
It is related to GCC RISC-V which changed their "march=xxxx" definition.
So either you need a old enough GCC version, either you need to update the makefile with -march=rv32im_zicsr  if i remember well.