SpinalHDL/VexRiscv #396: Debug instructions executed twice

patstew, 2024-03-06:
It seems like if you raise debug_cmd_valid less than 6 cycles after the previous instruction, the second instruction will be executed twice. This seems to be somehow related to IBusSimplePlugin_injector_decodeInput_valid being high, or the Fetcher otherwise passing the instruction on twice, but I'm finding it difficult to follow exactly what's supposed to be happening in there.

patstew, 2024-03-06:
And having written that last line, I now realise the problem is actually that the CPU is not getting halted when I remove the extra wait from the debug write task. I'll check what's going on, its a lot less surprising if the debug instruction is executed twice when the CPU isn't even halted :).

Dolu1990, 2024-03-06:
Ahhh XD
yes, that doesn't help ^.^

closed 2024-03-08 by Dolu1990