SpinalHDL/VexRiscv #400: Problems with adding FPU in Briey

15936866194, 2024-04-12:
Add FPU to Briey as follows
[image]
Compiled with the MABI=ilp32f MARCH = rv32imf compile option, pc points to the exception entry address when a floating-point instruction is encountered
The assembly code is as follows：
[image]
[image]
How can I solve this problem？

Dolu1990, 2024-04-12:
Hi,

Before using the FPU, in the software, you need to set the mstatus FS bits to something other than 00, else the FPU is disabled and will make things trap

15936866194, 2024-04-15:
Thank you for your answer. I used the generated.v file and vcs+verdi tool for simulation, but I am not very good at software Settings. What should I do for the problems you mentioned?
[image]
I found the fs description in FpuPlugin, can I manipulate it to fix the problem?

15936866194, 2024-04-15:
I'm sorry, I didn't understand what you meant before, but now I do. Let me try

Dolu1990, 2024-04-15:
Cool ^^

closed 2024-04-15 by Dolu1990