SpinalHDL/VexRiscv #401: Problem about how 通

ic-hjx, 2024-04-14:


Dolu1990, 2024-04-14:
Hi,

Maybe it is because the gcc packages are often compiled with a subset of architectural combination.
You may want to use rv32imaf instead, there is some chances it has it.
Else you can compile the gcc toolchain with your march specificaly : 
https://github.com/riscv-collab/riscv-gnu-toolchain

ic-hjx, 2024-04-14:
Thank you for the quick and helpful response to my issue; it's greatly appreciated! And I will try the method you mentioned right away.

> Hi,
> 
> Maybe it is because the gcc packages are often compiled with a subset of architectural combination. You may want to use rv32imaf instead, there is some chances it has it. Else you can compile the gcc toolchain with your march specificaly : https://github.com/riscv-collab/riscv-gnu-toolchain

ic-hjx, 2024-04-16:
I have recompiled the risc-v toolchain and compiled the program with `-march=rv32imaf_zicsr  -mabi=ilp32f`, fortunately the program can be successfully compiled now. 
But here comes the new problem that might be still caused by the FpuPlugin. After I connected to the briey SoC with OpenOCD and downloaded the program with GDB, the program with float type data couldn't run.
As you have said in the [#400](https://github.com/SpinalHDL/VexRiscv/issues/400), I have set the FS field as 2'b11 of mstatus in the crt.S.
```
crtInit:
  .option push
  .option norelax
  la gp, __global_pointer$
  .option pop
  la sp, _stack_start
  # set mstatus[13:14] FS field
  csrr x18, mstatus
  li x19, 0x6000
  or x18, x18, x19
  csrw mstatus, x18
```
And  in fact the program that I have used to test the FpuPlugin is simple, it's just sent the data with uart. While `float a=10.23;` is not defined, the program can run normarlly, but as long as the `float` appear in the code, the program will show anything.
```
int b = 12;
float a = 10.23;
myprintf("%d\n", b);
//myprintf("%f\n", a);

print("Hello !\n");
```

Thank you very much for your response again!

ic-hjx, 2024-04-17:
I typically connect to and debug FPGA development boards directly using OpenOCD and GDB. The core of the development board I use is the Xilinx xc7a100tcsg324. Previously, I indeed used GDB to directly troubleshoot the program, but now it seems I do need to conduct simulations to identify issues. I have written an additional top-level interface for briey to connect to external pin ports, and I will investigate how to simulate it.
Thanks for your advice!

ic-hjx, 2024-04-19:
I have been trying to simulate the Briey SoC behavior, and I have encountered some problem:
1.After I have run  `sbt "runMain vexriscv.demo.BrieySim"`, I can successfully connect the simulator with `openocd` and download the program with `gdb`. And after I run the program successfully, I can't find the wave file(.vcd) which should be dumped by the simulator.

2.Once I assign the `TRACE_INSTRUCTION` and `TRACE_REG` to be yes, the verilator simulation problem in `VexRiscv/src/test/cpp/briey` can't be compiled. Do I miss something to set up? Here is the error report:
```
../main.cpp: In member function ‘virtual void VexRiscvTracer::preCycle()’:
../main.cpp:51:25: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_arbitration_isFiring’
   51 |                 if(cpu->writeBack_arbitration_isFiring){
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp:52:71: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_INSTRUCTION’
   52 |                         instructionTraces <<  hex << setw(8) <<  cpu->writeBack_INSTRUCTION << endl;
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~
../main.cpp:56:25: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_RegFilePlugin_regFileWrite_valid’
   56 |                 if(cpu->writeBack_RegFilePlugin_regFileWrite_valid == 1 && cpu->writeBack_RegFilePlugin_regFileWrite_payload_address != 0){
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp:56:81: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_RegFilePlugin_regFileWrite_payload_address’
   56 |                 if(cpu->writeBack_RegFilePlugin_regFileWrite_valid == 1 && cpu->writeBack_RegFilePlugin_regFileWrite_payload_address != 0){
      |                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp:57:72: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_PC’
   57 |                         regTraces << " PC " << hex << setw(8) <<  cpu->writeBack_PC << " : reg[" << dec << setw(2) << (uint32_t)cpu->writeBack_RegFilePlugin_regFileWrite_payload_address << "] = " << hex << setw(8) << cpu->writeBack_RegFilePlugin_regFileWrite_payload_data << endl;
      |                                                                        ^~~~~~~~~~~~
../main.cpp:57:134: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_RegFilePlugin_regFileWrite_payload_address’
   57 |                         regTraces << " PC " << hex << setw(8) <<  cpu->writeBack_PC << " : reg[" << dec << setw(2) << (uint32_t)cpu->writeBack_RegFilePlugin_regFileWrite_payload_address << "] = " << hex << setw(8) << cpu->writeBack_RegFilePlugin_regFileWrite_payload_data << endl;
      |                                                                                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp:57:223: error: ‘class VBriey_VexRiscv’ has no member named ‘writeBack_RegFilePlugin_regFileWrite_payload_data’
   57 |  : reg[" << dec << setw(2) << (uint32_t)cpu->writeBack_RegFilePlugin_regFileWrite_payload_address << "] = " << hex << setw(8) << cpu->writeBack_RegFilePlugin_regFileWrite_payload_data << endl;
      |                                                                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
As I can see that the error is because verilator can't find the signals io about trace, I can find some wires about these signal in the VexRiscv module, but they weren't assigned to output io.

ic-hjx, 2024-04-22:
Thank you for your reply! After several simulations, I finally found that the problem originated from my incorrect setting of the FS field in mstatus. By directly modifying the 0x1808 to 0x7808 at [crt.S: line91](https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/9c1af28a9674b84991d3f4f004eef4c73090aaf6/resources/crt.S#L91) , the FPU core can work properly.
At the same time, by adding the FpuPlugin this time, I think the Plugin system in VexRiscv is really great. I would like to know if there is more detailed documentation available now for studying the Plugin system, or should I learn about the implementation of the Plugin system by reading the source code directly?

Dolu1990, 2024-04-22:
Ahhh nice :)

There isn't much documentation about the VexRiscv plugin system.
It is just a setup / build callback to handle things.

More evolved plugin system can be found in NaxRiscv / VexiiRiscv
Those ones are more documented :)

The VexiiRiscv one being the most recent iteration, it also made its way in SpinalHDL itself

ic-hjx, 2024-04-23:
OK, thank you for your nicely reply again! Now I think this problem can be closed.

closed 2024-04-23 by ic-hjx