SpinalHDL/VexRiscv #402: How to use printf function?

15936866194, 2024-04-15:
With the -nostdlib option used, the following error occurs
[image]
nostdlib is disabled and more errors are made

[ICer@IC_EDA /home/ICer/guochen/briey_fpu/briey_software/briey/uart]$make
/opt/riscv64-fpu/bin/riscv64-unknown-elf-gcc -march=rv32imf  -mabi=ilp32f  -o build/uart.elf build/src/main.o build/src/crt.o -march=rv32imf  -mabi=ilp32f -lgcc -mcmodel=medany -nostartfiles -ffreestanding -Wl,-Bstatic,-T,../libs/linker.ld,-Map,build/uart.map,--print-memory-usage 
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-puts.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-puts.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strlen.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strlen.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fvwrite.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fvwrite.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fwalk.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fwalk.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memchr.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memchr.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memcpy.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memcpy.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memmove.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memmove.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mlock.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mlock.o)
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-reallocr.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-reallocr.o)
[... cut ...]
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: build/uart.elf(.text): relocation ".L0 +0x0 (type R_RISCV_PCREL_LO12_I)" goes out of range
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-puts.o): file class ELFCLASS64 incompatible with ELFCLASS32
/opt/riscv64-fpu/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: 文件格式错误
collect2: error: ld returned 1 exit status
make: *** [../../resources/subproject.mk:46：build/uart.elf] 错误 1

Dolu1990, 2024-04-15:
Hi,

Can you compile https://github.com/SpinalHDL/VexRiscvSocSoftware/tree/master/projects/briey ?
Or this fail aswell ?

15936866194, 2024-04-16:
Hi，
I can compile https://github.com/SpinalHDL/VexRiscvSocSoftware/tree/master/projects/briey.
This problem arises when I use the "printf" function.

15936866194, 2024-04-16:
I'm sorry, I deleted something earlier, I'll try it.

Guochen-Shine, 2024-04-22:
I want to use  printf("a = %f\n",3.14);    to print floating-point data.
But I encountered the following problems:
[image]
I added the following code to stdlib.c
[image]
[image]
stdlib.c:(.text+0x6b0): undefined reference to `__truncdfsf2'
I think there is a soft floating point library missing double to float, what should I do?

Dolu1990, 2024-04-22:
Hi, Try with rv32imaf instead ? that one may be included in your toolchain.
Else you need to compile gcc yourself with specificaly rv32imf option.

Guochen-Shine, 2024-04-23:
Hi，thank you for your answer.I tried it like you said.But Error still exists.
[image]

Dolu1990, 2024-04-23:
i don't know then, that's compiler hell, probably the best would be to fine a better makefile than the one i had.

Guochen-Shine, 2024-04-24:
Thank you very much, I am trying to build my own gcc.

Guochen-Shine, 2024-04-24:
I tried to generate the new riscv32-unknown-elf-gcc.
https://github.com/riscv-collab/riscv-gnu-toolchain
../configure --prefix=/opt/riscv32 --with-arch=rv32imf --with-abi=ilp32f
I use the VexriscSoftWare default makeflie, modify the gcc path, to achieve the purpose.
[image]
Thank you for your suggestion.

Dolu1990, 2024-04-25:
Ahhh nice ^^

closed 2024-04-25 by Dolu1990