SpinalHDL/VexRiscv #411: Help for custom instruction

ztachip, 2024-06-08:
I try to create a custom instruction for fast context switching
The opcode will save a register value and at the same time restore the register to last saved value.
My code looks like below, it does not seem to work. I am doing this right?
Thx

  override def build(pipeline: VexRiscv): Unit = {
    import pipeline._
    import pipeline.config._

    // Custom opcode will save RS1 and restore last saved value to RD
    execute plug new Area {
      val rs1 = execute.input(RS1).asUInt
      val save_reg = Reg(UInt(32 bits))

      when(execute.input(IS_FAST_CONTEXT_SWITCH)) {
        execute.output(REGFILE_WRITE_DATA) := save_reg;
        save_reg := rs1;
      }
    }
  }

ztachip, 2024-06-08:
I got it to work with the code below. Can you confirm if I got it right? Thx

    execute plug new Area {
      //Define some signals used internally by the plugin
      val rs1 = execute.input(RS1).asUInt
      //32 bits UInt value of the regfile[RS1]
      val rs2 = execute.input(RS2).asUInt
      val rd = UInt(32 bits)
      val save_reg = Reg(UInt(32 bits))

      //Do some computations
      rd := save_reg;

      //When the instruction is a SIMD_ADD, write the result into the register file data path.
      when(execute.input(IS_SIMD_ADD)) {
        execute.output(REGFILE_WRITE_DATA) := rd.asBits
        when(writeBack.arbitration.isFiring) {
           save_reg := rs1
        }
      }
    }

ztachip, 2024-06-09:
I tried to convert the save_reg with a vector of save_reg so register saving are then pushed and poped as a FIFO
But it seems condition writeBack.arbitration.isFiring is not triggered sometime

Below is what I try to implement for fast register save/restore in context switch 
// Fast register save and restored, faster since no memory involved, register are saved as registers
SAVE_AND_RESTORE(x10,x10,x0) // Save register 10 and at the same time restore register 10 to last saved value
SAVE_AND_RESTORE(x11,x11,x0) // Save register 11 and at the same time restore register 11 to last saved value
SAVE_AND_RESTORE(x12,x12,x0) // Save register 12 and at the same time restore register 12 to last saved value
SAVE_AND_RESTORE(x13,x13,x0) // Save register 13 and at the same time restore register 13 to last saved value
:

What would be the right way that I can do this?
Thx

Dolu1990, 2024-06-10:
The issue with your implementation is that you may modify save_reg in a speculative manner, as the execute stage may be flushed by the branch plugin and the data cache.
So, the best is to move all your logic to the last stage of the pipeline.

Also you are mixing execute and writeBack stage, instead move all to writeBack stage ^^

ztachip, 2024-06-15:
Do you have some example on how to access writeBack stage for a custom instruction implementation.
For example how to I get reference to writeBack stage service class.
Thx

Dolu1990, 2024-06-21:
Hi,

One example for this could be the MulPlugin.scala, as it does things across all execution stages

https://github.com/SpinalHDL/VexRiscv/blob/207304727225c608c017130e244a8d8f479fa2f7/src/main/scala/vexriscv/plugin/MulPlugin.scala#L142
https://github.com/SpinalHDL/VexRiscv/blob/207304727225c608c017130e244a8d8f479fa2f7/src/main/scala/vexriscv/plugin/MulPlugin.scala#L48

Chaitanya-kumar-Y, 2024-07-21:
Hi sir, 
So i would like to know how are you able to solve this issue. Since i am also working on VexRISCV i got some doubts and facing some problems. If it is possible i would like to contact you for some guidance.

ztachip, 2024-07-21:
At the end, I did not need custom instructions so did not really pursue this

closed 2024-07-21 by ztachip