SpinalHDL/VexRiscv #414: how to send data serially using apbbus when using murax soc

karegoud, 2024-07-02:
iam trying using an ip which requires 16 inputs so instead of assigning different address for each inputs i wanted to send data serially and also receive 16 outputs  serially. can anyone help me with this data sending and receiving serially.

Dolu1990, 2024-07-09:
Hi,

I'm not sure to understand well.
What do you mean by "serially" ? do you mean UART ? inside / outside the FPGA ?
16 inputs => gpio ?

karegoud, 2024-07-12:
I am trying to add 64-point fft ip on murax soc using apb bus. As you know apb bus uses memory mapped registers and my input is complex so i have both real and imag part, same goes for output. so instead of assigning addresses to these many, can i send data serially using only one real port and imag port for input (similarly for output).

Dolu1990, 2024-07-15:
Sending things serialy to the same address ? Yes, you need to add a "fsm" at that address.

karegoud, 2024-07-17:
val processing: State = new State { 
      onEntry {
        
        index := 0
      }
      whenIsActive {
        vecin(7) := vecin(6)
        vecin(6) := vecin(5)
        vecin(5) := vecin(4)
        vecin(4) := vecin(3)
        vecin(3) := vecin(2)
        vecin(2) := vecin(1)
        vecin(1) := vecin(0)
        vecin(0).re := io.realin
        vecin(0).im := io.imagin
        index := index + 1
       when(index === 8) {	 
         goto(calculating)
        }
        
       I am using this code to store serially coming input to an array 
        
        is this correct and can you also tell me how i should send inputs serially in c code because when I am using for loop its not working for me .

closed 2024-07-17 by karegoud