SpinalHDL/VexRiscv #416: machineCsr test failing

jbrown11111, 2024-07-03:
I am running the machineCsr test using the verialted VexRisc and running into an error.

pc missmatch 80000020 should be 80000024
FAIL ../../cpp/raw/machineCsr/build/machineCsr at PC=80000020 I=34202e73
 REF PC=80000024 REF I=e1e63 time=4162

It seems like the verilated VexRisc is a step behind the reference VexRisc, or the reference VexRisc is a step ahead of the verilated VexRisc.  If I disable the reference VexRisc, I get another error

*** top->VexRiscv->lastStageRegFileWrite_payload_data is 2 but should be 15 ***

FAIL ../../cpp/raw/machineCsr/build/machineCsr at PC=80000020 I=34202e73 time=4350

Dolu1990, 2024-07-04:
Hi,

Which vexriscv config did you run ? because machineCsr will only pass if all the features are enabled.

jbrown11111, 2024-07-08:
yeah, I got it to pass.  I was using the "all" config of the csr, but I must have made some other changes to it the config that made things not pass.  When I tried again with default "all" config it was passing.

closed 2024-07-08 by jbrown11111