SpinalHDL/VexRiscv #418: new custom instruction in vexriscv

Chaitanya-kumar-Y, 2024-07-30:
I simulated the simd custom instruction in vexriscv. After that i try make the changes in the computations as given to see what changes i need to do in order to create a new custom instruction.But while simulating/testing with the same test file given in src/test/cpp/custom/simd_add, it is giving a error as attached .
[image]

`  override def build(pipeline: VexRiscv): Unit = {
    import pipeline._
    import pipeline.config._

    //Add a new scope on the execute stage (used to give a name to signals)
    execute plug new Area {
      //Define some signals used internally by the plugin
      val rs1 = execute.input(RS1).asUInt
      //32 bits UInt value of the regfile[RS1]
      val rs2 = execute.input(RS2).asUInt
      val rd = UInt(32 bits)

      //Do some computations

      rd(31 downto 0):= rs1(31 downto 0) + rs2(31 downto 0)
     //Instead of adding 8bits, i added all 32bits in single time just to see how the execution flow will go.
      //When the instruction is a SIMD_ADD, write the result into the register file data path.
      when(execute.input(IS_MAC_ADD)) {
        execute.output(REGFILE_WRITE_DATA) := rd.asBits
      }
    }
  }`

Dolu1990, 2024-07-31:
Hi, 
You would need to update https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/custom/simd_add/src/crt.S acordingly to the new instruction behaviour and compile it.

Chaitanya-kumar-Y, 2024-08-01:
Thanks for the help, it works. Now i want to create a non standard custom instruction format                            
                       | funct7  | rs3 | rs2 | rs1 | funct3 |  rd   | opcode |. So i would like to know which changes are should be made in decode and anywhere

Dolu1990, 2024-08-03:
Vex only support rs3 for the FPU.
If you dig in the other github issue/pr, i think there was one which was adding support for RS3 in the integer register file.

Else you could try VexiiRiscv, which would natively support RS3.

Chaitanya-kumar-Y, 2024-08-04:
Hi
I checked that github issue, so i made some changes in RegFilePlugin.scala and HazardSimplePlugin.scala like adding a RD register. Below is the code changes in RegFilePlugin file

`override def setup(pipeline: VexRiscv): Unit = {
    import pipeline.config._
    val decoderService = pipeline.service(classOf[DecoderService])
    decoderService.addDefault(RS1_USE,False)
    decoderService.addDefault(RS2_USE,False)
    decoderService.addDefault(RD_USE,False)
    
    decoderService.addDefault(REGFILE_WRITE_VALID,False)
  }`

`  
 val regFileReadAddress1 = U(shadowPrefix(srcInstruction(clipRange(Riscv.rs1Range))))
      val regFileReadAddress2 = U(shadowPrefix(srcInstruction(clipRange(Riscv.rs2Range))))
 	 val regFileReadAddress3 = U(shadowPrefix(srcInstruction(clipRange(Riscv.rdRange))))

      val (rs1Data,rs2Data,rdData) = regFileReadyKind match{
        case `ASYNC` => (global.regFile.readAsync(regFileReadAddress1),global.regFile.readAsync(regFileReadAddress2),global.regFile.readAsync(regFileReadAddress3))
        case `SYNC` =>
          val enable = if(!syncUpdateOnStall) !readStage.arbitration.isStuck else null
          (global.regFile.readSync(regFileReadAddress1, enable),global.regFile.readSync(regFileReadAddress2, enable),global.regFile.readSync(regFileReadAddress3, enable))
      }

      insert(RS1) := rs1Data
      insert(RS2) := rs2Data
      insert(RD) := rdData
    }

But when i try to compile this it gives error as 
`
not found: value RD_USE
[error]     decoderService.addDefault(RD_USE,False)

 not found: value RD
[error]       insert(RD) := rdData
`
So how to resolve this error

Dolu1990, 2024-08-05:
Where did you got RD_USE and RD from ?
shouldn't it be RS3 ?

Chaitanya-kumar-Y, 2024-08-05:
No, i am trying to implement a mac custom instruction 

`rd(31 downto 0) := rs1(31 downto 0) + rs2(31 downto 0) + rd(31 downto 0)`

Dolu1990, 2024-08-05:
Did you defined RD and RD_USE somewere ?
I mean, the error you have come from your code when scala compiles it, not SpinalHDL / VexRiscv one.

Chaitanya-kumar-Y, 2024-08-05:
I only defined RD and RD_USE in RegFilePlugin.scala like below

`decoderService.addDefault(RD_USE,False)` 

  `insert(RD) := rdData`

Used them in Hazardsimpleplugin and custominstruction files.  Should i define them somewhere ?

Dolu1990, 2024-08-05:
those aren't definition, those are usages

Check VexRiscv.scala, you will need to define them there

xavier-design, 2024-08-05:
Thanks alot. i defined them in VexRiscv, now it is working and i tested with the updated crt.S.

Dolu1990, 2024-08-05:
Cool ^^

closed 2024-08-05 by Chaitanya-kumar-Y