SpinalHDL/VexRiscv #419: Data buffer

xavier-design, 2024-08-04:
Hello,
I am working with a DSP block where i need to work  with multiple data. So I want to create a small type of buffer or cache to store all that data. That buffer/cache is fully dedicated to storing that particular data only. I want to create a custom instruction where i need to provide the base address of the buffer, rest all the data fetched starting from that base address and will do the operations. 
1.How to create that type of buffer/cache and assign the address to it?
2.Should i create two custom instructions where in first instruction i will fetch the data from that buffer and in the next custom instruction i will do the operations or is it possible in single custom instruction?

Dolu1990, 2024-08-05:
> How to create that type of buffer/cache and assign the address to it?

For this, use the writeback stage to only modify the state when things are sure to commit : 
writeback.arbitration.isFiring

> Should i create two custom instructions where in first instruction i will fetch the data from that buffer and in the next custom instruction i will do the operations or is it possible in single custom instruction?

Depend how much combinatorial path you have It is posible in a single instruction

xavier-design, 2024-08-05:
@Dolu1990 Thanks
I want to know, is it possible to create a custom instruction where i can fetch two memory data words 

`   custom(rd,r1,r2)   `
  ` rd = memory[r1] + memory[r2]`

r1 and r2 are more like base address
I know that at present it is designed to bring one memory data word, but i wish to know can we modify in a way such that it brings two data memory words either from data cache or main memory.

Dolu1990, 2024-08-05:
> I want to know, is it possible to create a custom instruction where i can fetch two memory data words

Not possible. VexRiscv is designed in a RISC manner, so no microcode.

xavier-design, 2024-08-05:
Thanks for the info @Dolu1990 
I am planning to implement a custom instruction which may do 3 or more multiplications and additions(so require multiple data words). So i am planning to assign certain part of data cache to be dedicated to store inputs and outputs of the operation. To design a custom instruction in which we gives the base address data cache where we stored the inputs, with offset bring multiple data words(not in a single go). I am confused that where should even i start? like changes in cache controller to bring data from/to that part of the data cache.

Dolu1990, 2024-08-13:
Hmmm can't do this kind of things, that would be realy big changes.
The thing which would be possible would be to added additional state (in addition to the integer register file)

xavier-design, 2024-08-21:
Okay thank you ,i will try that.

closed 2024-08-21 by xavier-design