SpinalHDL/VexRiscv #420: VexRiscv for custom processing in memory instructions

neha2351, 2024-08-07:
Hi ,

I am planning to customise the VexRiscv core and implement new plugin for processing in memory capabilities using one basic instruction lets say PIM_MEM_CPY which will copy the contents from the memory and paste it in the memory itself. Can this be done in a RISC way on this core?

Thanks

neha2351, 2024-08-09:
I plan to have a co-processor (PIM processor) in/near the memory for memory-intensive tasks. So the main task for this custom instruction will be to establish a communication between the CPU and the PIM processor to communicate to divide the memory intensive tasks and establish a handshake that way.

Dolu1990, 2024-08-16:
the CfuPlugin is kinda similar, probably a good one to look at as it fork/join instruction to streams