SpinalHDL/VexRiscv #422: Multiple register read from Register file

xavier-design, 2024-08-21:
Hello, so i have been working on reading  consecutive registers data into a custom simd instruction file , base starting from one of the source register file.
For example | funct7  | rs2 | rs1 | funct3 | rd | opcode |. Let rs1 = R1, rs2 = R4 , then i want to take rs1 = R1 as base register and then read R2,R3 registers data also. 
The changes i have made in register file and declared RS3, RS4 in Vexriscv.scala.

`
val regFileReadAddress1 = U(shadowPrefix(srcInstruction(clipRange(Riscv.rs1Range))))  
val regFileReadAddress4 = (regFileReadAddress1 + 1) & 0x1F
val regFileReadAddress5 = (regFileReadAddress1 + 2) & 0x1F
insert(RS3):= rs3Data
insert(RS4):= rs3Data
`

I read these to simd custom instruction to do operations by making changes in decoder service and enabling these registers. I have made changes in assembly program to test, but the tests showing error, what could be problem?

Dolu1990, 2024-08-23:
Ahhh very tricky 

I don't understand : 
insert(RS3):= rs3Data
insert(RS4):= rs3Data

Do you mean insert(RS4):= rs4Data ?

xavier-design, 2024-08-23:
Yes. i mean i am reading these new registers along with other registers like RS1 and RS2(not including full code,  only the changes).

insert(RS3):= rs3Data
insert(RS4):= rs4Data

Dolu1990, 2024-08-27:
So, no other way, realy need to do the same as for RS1 RS2

closed 2024-08-27 by Dolu1990