SpinalHDL/VexRiscv #424: Instruction that need Multiple cycles for execution

karegoud, 2024-09-01:
Hello @Dolu1990 can we have instruction that take multiple cycles for execution ?. If so can you tell me an example or how to achieve mutiple cycles for execution.

Dolu1990, 2024-09-01:
Hi,

Yes, the MulDivIterativePlugin does that for instance.

For instance via : 
https://github.com/SpinalHDL/VexRiscv/blob/919f00125d4fc76e7533eb23adec9d71fc62ae3f/src/main/scala/vexriscv/plugin/MulDivIterativePlugin.scala#L98C25-L98C31

karegoud, 2024-09-02:
In AES plugin I have seen usage of rom in memory plug area but what I am unable to understand is that contents of rom will be available at the end of clock cycle but data from rom was used in the same plug area(that means in the same cycle, logic is trying to use data before contents of rom is accessed). how is it possible .
[image]

I tried using ROM in my custom instruction in decode plug Area and wanted to use the contents of ROM in execute plug Area. so I thought of using stageable values for passing values from one stage to another but problem iam facing is when iam inserting rom contents value into stageable variable it is done at the start of cycle but actual required rom contents will be available at the end of clock cycle so wrong values are being sent to my execute stage . how to solve this problem.

Dolu1990, 2024-09-02:
Hi,

The trick is to use the address from one stage ahead (onExecute.romAddress) that way, the result comes when the instruction is in this memory stage

closed 2024-09-09 by karegoud