SpinalHDL/VexRiscv #426: DBusSimplePlugin's Wishbone support forces 32-bit reads

kivikakk, 2024-09-04:
I notice that `DBusSimplePlugin`'s Wishbone bridge forces all reads to appear to be 32-bits wide:

https://github.com/SpinalHDL/VexRiscv/blob/0255f51cc53be867a513a0ef3b088a6fced5d97e/src/main/scala/vexriscv/plugin/DBusSimplePlugin.scala#L167-L169

I'm not sure of the exact reason it was done this way, but I'm wondering if you'd be amenable to changing this default! For some MMIO it's helpful to be able to distinguish the read size.

Dolu1990, 2024-09-04:
Hmm i think you are right, this shouldn't be necessary to have this force to "1111"
Can you give a try without those 3 lines ?

kivikakk, 2024-09-04:
Yes, in fact I'm doing that on my fork, and it works perfectly well! I'll make a PR if you like?

Dolu1990, 2024-09-04:
Sure thanks ^^

PR #427 (2024-09-06): DBusSimplePlugin: don't force SEL to 1111 on read.
Closes #426.

Thanks! :)
closed 2024-09-06 by Dolu1990