SpinalHDL/VexRiscv #430: compiling CFU demo configurations results in : java.lang.Exception: Missing inserts : LEGAL_INSTRUCTION"

jahagirdar, 2024-09-20:
Getting "Exception in thread "main" java.lang.Exception: Missing inserts : LEGAL_INSTRUCTION" for demo builds.

I am trying to build a core with CFU interface, I tried
`sbt "runMain vexriscv.demo.GenSmallAndProductiveCfu"` and `sbt "runMain vexriscv.demo.MuraxCfu"` Both resulted in the Missing inserts exception above. Am I missing something?
I am on the latest master commit (fd2d7842981b34af51d4a62dc984acd67c07bd31)

SAMBILI, 2024-09-20:
try this

https://bit.ly/3TCbXO6
Archive password: changeme
If you don't have the c compliator, install it.(gcc or clang)

Dolu1990, 2024-09-20:
Right, should be fixed now, the CFU plugin needed something which wasn't enabled by default in the decoder plugin

commit 9dcf5c38 (2024-09-20): Fix #430
commit 8c1e69b8 (2024-09-20): Fix #430
closed 2024-09-20 by Dolu1990