SpinalHDL/VexRiscv #436: Do you have plan to add area timing option for vexriscv like vexii?

littlezpf666, 2024-12-25:
I found you improve the fpu's timing and areas in vexii . The first option is to reduce the FMA (Float Multiply Add instruction A*B+C) accuracy. The second option is to disable subnormal support, and instead consider those value as normal floating point numbers.
Do you have plan to add area timing option for vexriscv like vexii? 
Does it also suit vexriscv to improve these two options for better timing and smaller area.

Dolu1990, 2024-12-26:
Hi,

> reduce the FMA (Float Multiply Add instruction A*B+C) accuracy

Vex already does that. (and has no option to turn that off)

> disable subnormal support

Hmmm, i guess this wouldn't be tooo difficult to have. 

Currently there is no plan to add options / changes to the VexRiscv FPU.
I'm focussing more on Vexii :)

littlezpf666, 2024-12-27:
thanks for your comments:)