SpinalHDL/VexRiscv #437: Regarding the regression test...

nerdylye, 2024-12-26:
Hi, I'm a rookie in this, sorry if this is considered a dumb question.

First question:
I've ran the regression test according to the instructions provided: 
`export VEXRISCV_REGRESSION_SEED=42
export VEXRISCV_REGRESSION_TEST_ID=
sbt "testOnly vexriscv.TestIndividualFeatures"`

The tests passed, but does it have an output file for me to analyze what is being tested? 

Second question: 
May I know how does the regression test takes in the design file? What if I made a little modifications to the `smallest_nocsr` configuration and I would like to test it using the regression test, how should I do it?

Thank you!

Dolu1990, 2024-12-27:
> Hi, I'm a rookie in this, sorry if this is considered a dumb question.

No worries, that flow isn't well documented / nor pretty

> The tests passed, but does it have an output file for me to analyze the test results?

Right, all the logs are in the terminal you ran the command with (not nice).
There is another environnement variable to ask storing the waves of the simulations.

> May I know how does the regression test takes in the design file? What if I made a little modifications to the smallest_nocsr configuration and I would like to test it using the regression test, how should I do it?

Hmm to generate its verilog, be sure it goes into the root folder (of this repo)
Then you can cd to src/test/cpp/regression 

and : "make clean run REDO=10 IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=no MUL=no DIV=no  COREMARK=yes"

Not great XD

Dolu1990, 2024-12-27:
If you are interrested, there is VexiiRiscv, which has a better flow.

nerdylye, 2025-01-02:
Hii, thanks for the reply!

> Right, all the logs are in the terminal you ran the command with (not nice). There is another environnement variable to ask storing the waves of the simulations.

May i know what is the environment variable to store the simulation waveforms?

> Hmm to generate its verilog, be sure it goes into the root folder (of this repo) Then you can cd to src/test/cpp/regression
> and : "make clean run REDO=10 IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=no MUL=no DIV=no COREMARK=yes"

I generated the verilog using `sbt "runMain vexriscv.demo.GenSmallest"`, but I manually modify the `VexRiscv.v` verilog to add 1 extra memory. 
[image]

I tried running the simulation using the command you provided and it results in this error.
[image]

Any idea what is causing the error?

Dolu1990, 2025-01-17:
That error says that the hardware didn't behave the same way than the reference software RISC-V model.
REF => software model value written in the register file
DUT =>  what the hardware did
And it doesn't match.

closed 2025-04-22 by nerdylye