SpinalHDL/VexRiscv #439: How to add extra memory to proessor

karegoud, 2025-01-14:
I want to add additional memory in the VexRiscv for particular application and want to create custom instruction that can access this additional memory [address for which will be stored in the instruction]. Is there any plugin or example which an guide me.

Dolu1990, 2025-01-17:
Hi,

Hmm, there isn't a good example for something like that.
There is the AES custom instruction which embed some ROM to process things : 
https://github.com/SpinalHDL/VexRiscv/blob/7f2bccbef256b3ad40fb8dc8ba08a266f9c6256b/src/main/scala/vexriscv/plugin/AesPlugin.scala#L62

But nothing with writes into a ram.

karegoud, 2025-01-20:
I have reviewed the AES plugin, but it doesn't align with my needs. When VexRiscv performs load and store operations, it requires access to memory. How is this memory connected to the processor, and how can I add additional memory that I can access through my custom instruction? If you have any insights or suggestions, please let me know.

Dolu1990, 2025-01-21:
> When VexRiscv performs load and store operations, it requires access to memory

It access some external memory, this shouldn't be used as a reference as your case it to have very thightly coupled memory inside the CPU right ?

karegoud, 2025-01-22:
1. yes, I want tightly coupled memory inside CPU. Any references that might help me on how to add this tightly coupled memory[TCM] and can be used by custom instruction. 
2. How and where to setup the address space for this newly added TCM in Vexriscv
3. Is there any example in repository that has used data cache already present in the repo ?

Dolu1990, 2025-01-23:
There is no reference for that

One thing, maybe your memory should more be seen like a custom register file instead of a TCM ?