SpinalHDL/VexRiscv #440: Running Regression Test using post_synthesis verilog file

nerdylye, 2025-01-17:
Hii, 

May I know if the regression test can be performed on the post synthesis VexRiscv file? If can, how do I do it because I'm facing a lot of random errors haha...

Thanks!

Dolu1990, 2025-01-17:
Hi,

Hmm that was never done as far as i know.
There is now flow in place for it.

What kind of errors ? and on what synthesis tool ?

nerdylye, 2025-01-20:
Hii @Dolu1990,

I see, may I know how to use the flow?

The command that I'm running is `make clean run REDO=10 IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=no MUL=no DIV=no COREMARK=yes VEXRISCV_FILE=../../../../temp/VexRiscv_post_synthesis.v` in this directory `VexRiscv/src/test/cpp/regression`. 

I did include the primitives file in the Makefile code: `verilator -cc --trace ${VEXRISCV_FILE} ${PRIMITIVES} --top-module VexRiscv -O3 -LDFLAGS -pthread ${ADDCFLAGS} --gdbbt ${VERILATOR_ARGS} --Wno-IMPLICIT --Wno-UNOPTFLAT --Wno-WIDTH -Wno-fatal --noassert --x-assign unique --x-initial unique --exe main.cpp`

The error is this: 

[image]

nerdylye, 2025-01-22:
Hii @Dolu1990, thanks for the swift reply :D!

I've tried using `VexRiscv.v` for the post synthesis verilog file, but its still giving the same error. 

> Hmm that was never done as far as i know.
There is now flow in place for it.

You mentioned above that there is now a flow for running post synthesis tests on VexRiscv, could you share more about it?

Dolu1990, 2025-01-22:
Hoo i meant "There is no flow in place for it." XD sorry.

For the error you had, check what file were generated by verilator, that is what the testbench can't find apparently.

closed 2025-03-03 by nerdylye