SpinalHDL/VexRiscv #443: Partitioning On-Chip RAM in Murax SoC and Mapping IP Access

xavier-design, 2025-02-12:
I want to partition the on-chip RAM of the Murax SoC into three distinct regions:
6 KB General Memory: Accessible by the CPU for standard operations.
1 KB Application-Specific Memory: Dedicated to a hardware IP (e.g., matrix multiplication) for read/write access to application data.
1 KB CPU-Exclusive Memory: Reserved for CPU read/write operations.
Questions::
1.How should the on-chip RAM be partitioned?Should the partitioned memory regions be connected to the existing memory bus, or should a separate bus be created for isolated access?
2. How can a peripheral IP and CPU access its dedicated memory region?What is the recommended approach to map the IP and CPU to this specific memory space?
3. Which files should I review for better understanding of on-chip memory access and memory access?

Dolu1990, 2025-02-12:
Hi,

> or should a separate bus be created for isolated access?

So, the CPU will eat a lot of bandwidth as it has no cache => you will need to have dedicated ram for the hardware IP.
Or maybe you need to enable the I$ in the CPU.

What are the difference between : 
- 6 KB General Memory: Accessible by the CPU for standard operations.
- 1 KB CPU-Exclusive Memory: Reserved for CPU read/write operations.
 ?

The hardware IP can also access the 6KB general memory ?

One important thing to keep in mind : 
- Let's say, you have the CPU and the hardware IP which access memory through a shader bus. 
- That shared bus goes to two memories
- Only one memory can have pending accesses at a given time

This is because the bus decoder only allows 1 peripheral to have memory request at the time, that ensure the proper ordering of memory responses.

xavier-design, 2025-02-12:
Hello @Dolu1990 , thank u for the fast response.

> you will need to have dedicated ram for the hardware IP.

Yes thats why the parition of onchip ram.

>What are the difference between :
>6 KB General Memory: Accessible by the CPU for standard operations.
>1 KB CPU-Exclusive Memory: Reserved for CPU read/write operations.

The two 1 KB memory regions facilitate data exchange between the CPU and the hardware IP. The CPU Memory Region (1 KB) allows only the CPU to read and write, while the IP can only read from it. Conversely, the IP Memory Region (1 KB) allows only the IP to read and write, while the CPU can only read from it. This is to ensure controlled and isolated data transfer.

>The hardware IP can also access the 6KB general memory ?
No, the hardware IP can only see 1KB which it can read and write, and 1KB(CPU region) from which can read only.

>Only one memory can have pending accesses at a given time
Yeah, the memory should be accessed by either CPU or IP at a time.

Dolu1990, 2025-02-17:
> How should the on-chip RAM be partitioned?Should the partitioned memory regions be connected to the existing memory bus, or should a separate bus be created for isolated access?

Create isolated bus as much as possible

> How can a peripheral IP and CPU access its dedicated memory region?

I don't understand the question.

> What is the recommended approach to map the IP and CPU to this specific memory space?

The only think i would recommand is to not map anything at address 0x00000000 to avoid null pointer having memory.

> Which files should I review for better understanding of on-chip memory access and memory access?

?