SpinalHDL/VexRiscv #445: Inquiry on running Dhrystone benchmark flow on VexRiscv

nerdylye, 2025-03-04:
Hi @Dolu1990, the Dhrystone benchmark on this repo is already in the form of .hex files in `/src/test/resources/hex` which is specifically made to test VexRiscv cores. 

I am looking for a flow in which I could run Dhrystone benchmark on multiple riscv cores using the same flow. I'm currently exploring the picorv32 Dhrystone flow but it doesn't seem to be able to run the test on VexRiscv. 

May I know in your knowledge, is there a general flow out there that I could use to run Dhrystone benchmark on VexRiscv and picorv32 cores to compare their performance?

Any help is much appreciated, thank you!

Dolu1990, 2025-03-04:
Hi,

Each CPU has his own testbench setup/layout. None are compatible with each other in general, so each one need its own dhrystone binary.

Note there is also VexiiRiscv dhrystone : https://github.com/SpinalHDL/NaxSoftware/tree/c8e49bd15d60948c8a272e9c28052ea6ef1e2c46/baremetal/dhrystone_vexii

nerdylye, 2025-03-05:
Thanks for the swift reply @Dolu1990, but I don't quite understand, could you explain a little more please?

From what I know, the Dhrystone benchmark usually consists of `dhrystone.c`, `dhrystone.h` and `dhrystone_main.c` as their main files. The riscv gcc compiler will take these files and convert into binary `dhrystone.bin`. In theory, all that's left is to load the binary into the riscv cpu and it will run the benchmark. Since the 3 main dhrystone files are the same, shouldn't the compiled binary to be loaded to the cpu also same?

Dolu1990, 2025-03-05:
Aren't the same fully, because those are compiled to run on baremetal (no OS), so they need to have specific access (function of the target) to a time reference and putchar functions.

closed 2025-04-21 by nerdylye