SpinalHDL/VexRiscv #446: Vexriscv dual-core smp architecture has a problem when debugging

TieTouXiaoGe, 2025-03-11:
Hi @Dolu1990, 
I'm currently using this demo:
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexCluster.scala, Dual-core configuration.
Also, I'm using the official debug, so privilegedDebug is true. And then I changed this: https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpCluster.scala#L139, To replace it into "DebugTransportModuleJtagTap". Because I want to generate a jtag interface that is a standard four-wire in verilog.
I have transplanted the generated verilog to my FPGA board and run a lot of software normally. Recently, I want to experiment the debug function, so I constrained four JTAG wires to go out and debug a single core respectively, which is normal. I used gdb+openocd+ftdi combination.
When debugging dual core, that is, openocd config file using "target smp xxx" command configuration (I'm sure my config file is OK, it's generic), gdb command display thread, switch thread is fine. The problem is: for example, hart0 performs single-step debugging first, when switching to hart1 for single-step debugging, the program is not executed line by line, but directly jumps to the stop position of hart0. I feel that the dpc value of hart1 follows hart0, and it seems not updated.
Looking forward to your reply. This problem has bothered me for a long time. Thank you very much.

Dolu1990, 2025-03-14:
Hi,

I wasn't aware of the bug.

Can you provide a way to reproduce ?
In particular, there is ways to make a simulation of it (including jtag) via https://github.com/enjoy-digital/litex/pull/1887

TieTouXiaoGe, 2025-03-15:
Thank you for your reply. Maybe I didn't express it clearly.
1. For a CPU with a dual-core SMP architecture, you can debug a single hart separately, that is, create a single target, core0 or core1, in the openocd configuration file.Use the "info thread" command to display only one thread.In this case, I have tested them separately, and it is normal.Now the problem is to debug the dual-core SMP type of software by creating two targets in openocd and using the "info thread" "command to show two threads.The reset PC value is now the same for both harts.In theory, stepping through both harts is performed instruction by instruction, that is, PC + 4 goes on like this.At present, when I use "thread 1"to select hart0 for debugging, it is normal to debug hart0 in a single step. It is executed one by one. For example, if I stop at PC + 0xC in a single step, I use"thread 2" to switch to hart1. The single step debugging directly jumps to the PC + 0xC address where hart0 just stopped.Rather than executing the instructions one by one.
2. The simulation method you recommend, I'll look into it, I haven't used it yet.
3.  I changed this: https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpCluster.scala#L139, To replace it into "DebugTransportModuleJtagTap". Because I want to generate a jtag interface that is a standard four-wire in verilog. Can I modify it in this way without affecting the debugging under the dual-core SMP architecture?I'm sorry because I'm not even familiar with Verilog. I'm a software developer.Because of the ease of use of your great spinal HDL, I was able to make a few small changes.

TieTouXiaoGe, 2025-03-15:
As for how to reproduce,I use https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexCluster.scala.Modify the debug place, and then compile the generated Verilog file, with my peripherals: memory and peripheral modules, to form a complete SoC system, which can be downloaded to the FPGA development board.

Dolu1990, 2025-03-16:
Hi,

> DebugTransportModuleJtagTap

That should be all good, no worries on that side.

> reproduce

I would need the openocd scripts you use, and know which exact version of open ocd you are using.

The closest my setup is, the better

TieTouXiaoGe, 2025-03-17:
Ok, openocd version: 

[image]

openocd scripts:

- adapter speed 1000
- 
- set _CHIPNAME riscv
- jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10002fff       
- 
- set _TARGETNAME $_CHIPNAME.cpu
- 
- target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME -rtos hwthread
- target create $_TARGETNAME.1 riscv -chain-position $_TARGETNAME -coreid 1
- target smp $_TARGETNAME.0 $_TARGETNAME.1
- 
- riscv set_command_timeout_sec 30
- 
- init
- if {[ info exists pulse_srst]} {
-   ftdi_set_signal nSRST 0
-   ftdi_set_signal nSRST z
- }
- halt

TieTouXiaoGe, 2025-04-25:
Hi,Can you reproduce it over there?

TieTouXiaoGe, 2025-04-27:
Ok, I'll try.