SpinalHDL/VexRiscv #448: Porting FREERTOS to VexRiscv

nerdylye, 2025-04-21:
Hi @Dolu1990, 

I found this blog `https://hackmd.io/@4a740UnwQE6K9pc5tNlJpg/H1olFPOCD#Reference-list` on how to run FREERTOS on VexRiscv as the current repo does not have any information regarding this. I followed his steps but I'm still getting the `pc missmatch` error, do you happen to know why?

[image]

The ultimate goal is to run FREERTOS on MuraxSOC/BrieySOC. 

Thank you!

Dolu1990, 2025-04-23:
Probably the VexRiscv config you are using has some CSR disabled that freertos needs ? checl what is at the REF PC=xxxx instruction in the assembly.

nerdylye, 2025-04-24:
Thanks for the swift reply @Dolu1990 , i found the problem. 

One question: Since the regression test loads the HEX files of each test in `/src/test/cpp/regression/main.cpp`. May I know how the HEX files of each test is generated and where can I access the source codes for the tests? Because for MuraxSoc, there is an external repo [https://github.com/SpinalHDL/VexRiscvSocSoftware](url) to generate the HEX files for the test using riscv32-unknown-elf-gcc tools, I was wondering if there is the same thing for the tests ran on VexRiscv CPU core...

Dolu1990, 2025-04-25:
> May I know how the HEX files of each test is generated and where can I access the source codes to generate the HEX file for the tests?

It was a mess, very long time ago.

Probably some modifications done in https://github.com/Dolu1990/FreeRTOS-RISCV
Very outdated stuff.

nerdylye, 2025-04-28:
I managed to compile the Dhrystone files from VexRiscvSocSoftware and ran on MuraxSoC, it was working. But when I tried to run the same Dhrystone HEX file on just the VexRiscv (smallestnocsr) config with this command `make clean run IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=no MUL=no DIV=no DHRYSTONE=yes`, it has this PC mismatch error. Is it because of the the dhrystone source files for MuraxSoC is not compatible with VexRiscv smallestnocsr core?

[image]

Dolu1990, 2025-05-01:
Right, Murax software isn't compatible with the SoC less testbench.