SpinalHDL/VexRiscv #449: Wierd DBus / Register File behavior when implementing DCache & FPU. The register file write index is mixed with DBus Write on dready_i

wye1102, 2025-04-23:
I'm trying to implement FPU with DCache. The bug occurs always with a sequence of lw, branch, sw, lw, lw ... addi, ret. The branch is not to be executed. The sw inst. after branch inst. is incorrectly triggering a register write with ra, and DBus data, causing ret inst. to enter abnormal states. The case is witnessed in my scala configuration regardless of earlyInjection settings & dcache settings. 

Setting DBusCachedPlugin_mmuBus_rsp_isIoAccess = 1, reducing peripheral request latency may help with some situations, however it'll always stuck at some certain instruction sequences. After a deep debug I've found that the dready_i of sw inst. after branch inst. will overwrite ra value, and register write seems to be advanced by 1 cycle earlier. 

I have attached a sample scala configuration and a piece of simulation data. Hopefully to find a solution for this problem. RV32IM can be too slow when a lot powf & expf insts. are used. 

Thanks.

The configuration scala file
[GenTest.zip](https://github.com/user-attachments/files/19872317/GenTest.zip)

The code asm file
[boote.zip](https://github.com/user-attachments/files/19872349/boote.zip)

IBus & DBus RW (code at 64f0 to 6514). 
[image]

Accidental RegFile Write. From the image ra actually written to s0, s0 to s1, etc. 
[image]

Modelsim simulation waveform data
[Test.zip](https://github.com/user-attachments/files/19872554/Test.zip)

Dolu1990, 2025-04-25:
Hi,

> DBusCachedPlugin_mmuBus_rsp_isIoAccess

What is your io region configuration ?
Is the FPU accessing the io region ?

Just to understand the situation.

wye1102, 2025-04-25:
I've done several tests, and finally makes the FPU run. There're several keys to follow: 

1. The FPU can only access to cached memory, otherwise failure may occur at anywhere. 
2. When altering mcause the FS flag must be kept or set to DIRTY  otherwise the FPU will simply stopped working !! This costs me several hours to look into until I tracked the FPU cmd valid flag to hold low. 

There is still a problem concerning the readback data from io_mem_rsp port. When performing back-to-back write->read at IO range (memory copy), the first read data is accidentally written back to register file, instead of the expected subsequent read data. The read data at R Data is correct, however the memory write data is always the first data (49b9e7a7). Here is a sample timing diagram: 

[image]

The W indicates IO Write, and R indicates IO read. The operation is read to write. The read command is delayed until io_mem_cmd_ready. W request is asserted 2T after execute stage. However the writeBack stage seems to take data along with io_mem_cmd_valid for the read command. 

I have tested 0T memory model and this 1T model, both behave the same way... The computer cannot get the correct data. However when doing loop copy for the whole packet, the data is correct. I'll try to collect more data then.

Dolu1990, 2025-04-26:
> The FPU can only access to cached memory, otherwise failure may occur at anywhere.

Right, i think it was by design, not as a bug.
Maybe to make it clean, the CPU should trap instead.

> When altering mcause the FS flag must be kept or set to DIRTY otherwise the FPU will simply stopped working !! This costs me several hours to look into until I tracked the FPU cmd valid flag to hold low.

So, you mean that if FS is set to 0, the FPU stop working ?
This is the intended RISC-V behaviour specified by the spec. I understand the frustration XD

> There is still a problem concerning the readback data from io_mem_rsp port. When performing back-to-back write->read at IO range (memory copy), the first read data is accidentally written back to [...]

load/store done via integer register file ?

> Data cache is set to 4 bytes per set

Ahhhh, don't expect Vex to support this. This is out of scope, was never tested / designed to support this corner case configuration.

wye1102, 2025-04-27:
Yeah the ld/st is done via integer register file at the I/O range. The dataCache module does not seem to work properly with back-to-back instr feeding and / or data feeding in my SOC implementation. 

Have tested several optimizations but access to I/O range is always buggy. Cached range hasn't been fully identified, however the simulated UART report is incorrect. IBUS is using simple model, and is also performing back-to-back reads. 

The 4 byte per set data cache is currently functioning properly with RV32IMAF configuration in VexRiscv. 

Finally makes the core working by adding some bubbles in instr feeding pipeline. Adding 4T latency for 5 instructions after LD/ST now eliminates the bug. 

I'll try later with instr cache enabled (area is a little bit critical). Thanks for the help.

wye1102, 2025-04-27:
Did a simple test with ICache enabled, and both ICache & DCache have a 4-byte line length. Showed the same issue: 

[image]

Look at the back-to-back read->write sequence at IO range. The ICache is stuck normally for several intervals, however the read data is not properly write back into register a5. Here is the source asm: 

[image]

When using IBusSimplePlugin, the above issue can be eliminated by inserting bubbles in IBus feeding pipeline, and the processor works properly (all APIs are almost fully tested) however slower. I'll try increase cache line to 8 bytes and test again.

wye1102, 2025-04-28:
Hi I've created a fast AXI wrapper and have tested every software API using a regenerated DAxi core. Everything seems to work fine now finally. 

The DAxi will automatically insert bubbles for Writes since it's write-through cache, and the AXI controller has to wait for BVALID. I'll fully switch to DAxi implementation for high performance cores. The native DBus will only be used without FPU and cache (small). 

Thanks again for the help.