SpinalHDL/VexRiscv #451: Integrating vhdl using Blackbox

karegoud, 2025-06-04:
am trying to integrate a VHDL file using a BlackBox in SpinalHDL. I used addRTLPath(s"./rtl/myDesign.vhd") to include the file, but I’m still getting an error in Verilator that says **the module definition of myDesign is not found in Murax.v.**

Also, is it correct to add the BlackBox instantiated module as a peripheral in the Murax SoC?

karegoud, 2025-06-06:
even with verilog file iam getting same error

code for blackbox:
package vexriscv.periph.xxxx
import spinal.core._
import spinal.lib._

class Multiplier extends BlackBox {
  val io = new Bundle { 
        val clk = in Bool() 
	val reset = in Bool()
     .........
   }
   
   
   ClockDomainTag(this.clockDomain)(io)  // applying tag to whole Bundle 
   
   noIoPrefix()
  
  mapCurrentClockDomain(clock = io.clk)  //, io.reset
  addRTLPath("home/ubuntu/......../Multiplier.v")  
}

code for instantiating blackbox:

package vexriscv.periph.xxxx
import spinal.core._
import spinal.lib._

case class TopLevel() extends Component {
	
	val io = new Bundle { 
	val clk = in Bool() 
	val reset = in Bool()
	........
   }
   
   
   val mul = new Multiplier()  // instantiating blackbox
   
   // connecting all signals
   io.reset <> mul.io.reset
  .......
}

object Main {
	def main(args: Array[String]): Unit = {
	SpinalVhdl(new TopLevel)
	}
}

Can you please tell me what am I missing, that is leading to error

karegoud, 2025-06-11:
Code that instantiate Blackbox:
```
case class TopLevel() extends Component {
	val io = new Bundle { 
	val clk = in Bool() 
	val rst = in Bool()
	val valid = in Bool()
	val a = in (Bits(8 bits))
	val b = in(Bits(8 bits))

        val p = out(Bits(16 bits))
	val ready = out Bool()
   }
   
   val mul = new Multiplier()  // instantiating blackbox
 
   io.valid <> mul.io.valid
   io.a <> mul.io.a
   io.b <> mul.io.b
   io.p <> mul.io.p
   io.ready <> mul.io.ready
  
}

object Main {
	def main(args: Array[String]): Unit = {
	SpinalVhdl(new TopLevel)
	}
}
``` 
Code for Blackbox:
```
class Multiplier extends BlackBox {
  val io = new Bundle { 
	val clk = in Bool() 
	val rst = in Bool()
	val valid = in Bool()
	val a = in (Bits(8 bits))
	val b = in(Bits(8 bits))

        val p = out(Bits(16 bits))
	val ready = out Bool()
   }
   
   
   ClockDomainTag(this.clockDomain)(io)  // applying tag to whole Bundle 
   
   noIoPrefix()
 
  mapCurrentClockDomain(io.clk, io.rst)  //, io.reset (clock = io.clk)
  
  addRTLPath("/home/ubuntu/MTP-I/VexRiscv/src/main/scala/vexriscv/peripheral/IP_blackbox/Multiplier.v")  
}
```

Instead of VHDL file I used verilog file to integrate. If you need I can also share verilog file.
In the generated RTL, the top-level module includes an instantiation of the BlackBox. However, the Verilog definition of the BlackBox module itself is missing from the final output.
For example, the top-level connects to the Multiplier BlackBox, and I can see the instance in the generated Verilog. But the actual module Multiplier (...) definition is not present anywhere in the output files.
I used addRTLPath("path/to/Multiplier.v") inside the BlackBox class, expecting that the module definition would be included automatically.

Could you clarify why the BlackBox Verilog module isn't being copied or linked into the output, and whether I need to manually handle that?

Dolu1990, 2025-06-17:
Ahhhh it is normal, this addRtlPath is just to document things (and be used by the tools downstream), not to change any hardware generation.

Basicaly, when you can get the list of all the RTL files via : 

```scala
val report = SpinalVhdl(new TopLevel)
report.rtlSourcesPaths.foreach(println)
```